kerberos-io/machinery #2: Stopped working after upgrade ("pacman -Syu") ("pacman -Syyu").

dossjb, 2015-06-03:
I have tested kerberos on a Raspberry pi B+ and a Raspberry Pi 2 B+.  Both seemed to be working great until I attempted to upgrade them using the "pacman" package manager.

cedricve, 2015-06-03:
Can you create a ticket on https://kerberosio.zendesk.com/hc/en-us. We would like to resolve your issue, as we want to know what exactly is going wrong.

closed 2015-06-04 by cedricve

dossjb, 2015-06-05:
Today I checked my Kerberos pi and everything seemed to be working okay.  Not sure what changed.  The other day when I checked it after the upgrade, it showed no images for both of my new and old B+ pis.

cedricve, 2015-06-05:
We have some caching on the filesystem, because I/O operations on the Raspberry Pi (read SD card) are slow. That could be a reason why you didn't saw any changes, because it was still cached. I don't think this has to do something with upgrading the Arch Linux environment, we just released another version 1.0.2, which includes additional fixes and also an upgraded/updated Arch Linux environment.