kerberos-io/machinery #4: IP camera support

cedricve, 2015-06-29:
Create a new capture device `IPCamera'.

cedricve, 2015-06-29:
IP Camera #3

cedricve, 2015-07-30:
IP camera is available on **develop branch** and can be tested with a MJPEG endpoint. E.g. you can use your mobile phone and use an IP camera app (https://play.google.com/store/apps/details?id=com.pas.webcam&hl=nl).

Change Capture device to **IPCamera**, and define the URL of the ip camera with it's resolution. An example for the IP camera app would be:

Url: 192.168.0.100/shot.jpg
Width: 800
Height: 600

stephenjamieson, 2015-08-03:
Looks like I cannot test this since my ip camera does not have an mjpeg feed, only h264.

cedricve, 2015-08-04:
According the OpenCV documentation, I think it should work.

stephenjamieson, 2015-08-04:
The only feed that the camera provides is an rtsp stream. I tried to force it into "mjpeg" mode but it doesn't work right.  I think I would need something to split out the stream into something useable, which could probably be done with ffmpeg, but I'm not sure how

cedricve, 2015-08-04:
Did you tried it with the 264h feed? According to the current OpenCV documentation, it should work with ffmpeg now. In previous release it was broken.

To update your current installation execute following commands

pacman -S ffmpeg curl
cd /home/kerberos-io
git pull
git checkout develop
cd build
make

After the compilation is done reboot the RPi, and change the configuration on the web interface.

stephenjamieson, 2015-08-04:
I did try the `rtsp://user:user@ip/videofeed ` uri that works with vlc, but nothing happens with kerberos, I'm not seeing anything in the logs so I'm not sure what's happening at all.

cedricve, 2015-08-04:
You recompiled and install ffmpeg, just like the commands I've described in previous command?

stephenjamieson, 2015-08-04:
I did yesterday, I don't see any commits of relevance since then

cedricve, 2015-08-04:
Can you force the rtsp on TCP?

stephenjamieson, 2015-08-04:
Does not look like it, tries udp first and fails to tcp

cedricve, 2015-08-04:
I would like to get this problem solved, can you join gitter? https://gitter.im/kerberos-io/hades?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge

stephenjamieson, 2015-08-04:
On ubuntu I installed 

```
ffmpeg libgstrtspserver-0.10-0 libgstreamer-plugins-base0.10-dev  libswscale-ffmpeg-dev libswresample-ffmpeg-dev libavresample-ffmpeg-dev libavcodec-ffmpeg-dev 
```

for my foscam:
appended ?tcp to the url (rtsp://user:pass@ip/videoMain?tcp)

and magic!

Thanks @cedricve confirmed working

stephenjamieson, 2015-08-06:
Still seems to crash, I added the `captureDelayTime=5` but the feed soon goes black and I have to reset the device.  Without kerberos it runs fine, and vlc streams it fine.

cedricve, 2015-08-06:
Thank you for testing, I will investigate this further.

cedricve, 2015-08-06:
@stephenjamieson the capture delay time is in miliseconds, can you try 5000 e.g.?

stephenjamieson, 2015-08-06:
I hasn't crashed for about 5 minutes since changing to 5000, I'll keep an eye on it

cedricve, 2015-08-07:
If it worked, can you change the value to 500-1000 and see what happens?

cedricve, 2015-08-13:
@stephenjamieson 

Did you installed gstreamer?

    sudo apt-get update
    sudo apt-get install libgstreamer0.10-0 libgstreamer0.10-dev gstreamer0.10-tools gstreamer0.10-plugins-base libgstreamer-plugins-base0.10-dev gstreamer0.10-plugins-good gstreamer0.10-plugins-ugly gstreamer0.10-plugins-bad gstreamer0.10-ffmpeg

maco1717, 2015-08-14:
Dude! I haven't had a chance to try this yet I'm looking forward to get some time try this master piece. does is support storing on a share/NAS? that's the only thing I was apart from supporting IPCams that you've already taken care of. I wouldn't think is to difficult, just mapping the share on linux as a drive and communicating to you app to store there, but as I said haven't even had a change to try your app all speculations ATM. kind regards an congratulations.

cedricve, 2015-08-14:
@maco1717 yes, follow this document https://doc.kerberos.io/develop/addons/NAS. You'll need to make some changes on the Raspberry Pi.

cedricve, 2015-09-12:
works for mjpeg streams

closed 2015-09-12 by cedricve

mcgyver83, 2017-01-04:
Hi all, I'm running kerberos.io on a raspberry pi2 but I cannog get my Sricam ONVIF ipcam working with RTSP streaming.
This is working with vlc and rtsp://192.168.0.101:554/onvif1 but I cannot get it working with kerberos :(

```
04/01/2017 22:03:48.714 INFO  [trivial] Starting capture device: McG-camera
04/01/2017 22:03:48.715 ERROR [trivial] Kerberos : Factory : Could not create an instance of message: McG-camera
```

```
<streams>
    <Rtsp>
        <enabled type="bool">true</enabled>
        <streamPort type="number">554</streamPort>
        <quality type="number">75</quality>
    </Rtsp>
</streams>

<captures>
    <IPCamera>
        <url type="text">rtsp://192.168.0.101:554/onvif1</url>
        <frameWidth type="number">640</frameWidth>
        <frameHeight type="number">360</frameHeight>
        <delay type="number">1000</delay>
        <angle type="number">0</angle>
    </IPCamera>
</captures>
```