kerberos-io/machinery #9: Video instead of single jpg's

stephenjamieson, 2015-07-06:
Hello, I've been testing this out on a VM and so far I'm impressed.  The one thing this project is missing is the ability to record video and not just single jpegs.  Not sure if this is feasible, but it would be great for things like moving cars that are too fast to catch on one or two frames.

cedricve, 2015-07-07:
That would require a complete different setup, as Kerberos would need to continuously record video. Afterwards Kerberos would need to take samples from the video feed, and mark the timestamps when motion was detected. For visualization we need, similar to Dropcam, to have a timeline which marks the amount of events happened at a specific point in time.

What are your thoughts on this? Please keep in mind that we can't store a huge amount of video storage on the SD card.

stephenjamieson, 2015-07-07:
This sounds what I'm looking for, but like you say it may be difficult for a Raspi with limited resources.  This obviously could be fixed with your cloud product, or some other external storage mechanism.  On my full fledged server this isn't an issue.

cedricve, 2015-08-04:
We will not implement this feature for the moment. As the implementation will takes us to far, additional solutions exists in combination with IP camera software.

closed 2015-08-04 by cedricve