kerberos-io/machinery #12: Raspberry Pi Camera dark images

cedricve, 2015-07-27:
Raspberry Pi camera shows dark images and doesn't operate at night.

kr15h, 2015-07-27:
I believe that is not a bug. There are 2 types of cameras for the Pi:

1. the normal one
2. the NOIR one (without IR filter)

You won't be able to see in the dark with the normal one. With the NOIR the trick is that it can not see in the dark just like that. You need an infra-red light source. As infra red light rays are not visible by the human eye, we perceive the IR light as invisible, but the Raspbery Pi camera is sensitive to it, thus recording the IR rays that are reflected.

cedricve, 2015-07-29:
That's true kr15h, but the main problem with the Raspberry Pi camera is that the images taken by the camera are very dark compared to the results of the raspistill command.

kr15h, 2015-07-29:
Ok, clear. that's a matter of settings then most probably. What library do you use for taking the pictures? You could use the same `raspistill` from within C++ code. Or that would be too slow?

cedricve, 2015-07-29:
Indeed, I've been using the source of AVA RaspiCam but made some small modifications for OpenCV 2.4. You can find it here https://github.com/cedricve/raspicam. This library wraps the userland library of Raspberry Pi. I didn't had the time to spend more time on finding this issue, as the source is not very userfriendly.

FranciscMoldovan, 2016-03-11:
https://github.com/cedricve/raspicam/issues/2#issuecomment-195032172

FranciscMoldovan, 2016-03-11:
The frame rate is at a constant 30 fps, as can be seen in the highlighted line of code below:
https://github.com/cedricve/raspicam/blob/master/src/private/private_impl.cpp#L71 
30 fps is the reason why manual exposure is only mapped on the interval [0,33] ms, because each frame can have an exposure (be it manual, or automatic) of 1/30 s, that is 33ms.
Solution for better exposure: change in the line of code from the link above, the fps:
from
`State.framerate 		= 30;`
to 
`State.framerate 		= 10;`
Next, recompile the library, by going to the build directory, and inserting the commands from the tutorial

https://github.com/cedricve/raspicam#compiling

You will have a frame rate of 10 fps, but can obtain an exposure as low as 100ms (in the case of 10 fps)

below is the code I've used for having a preview on the raspberry of the exposure range

```
#include <raspicam/raspicam.h>
#include <raspicam/raspicam_cv.h>
#include <opencv2/core/core.hpp>
#include <opencv2/imgproc/imgproc.hpp>
#include <opencv2/objdetect/objdetect.hpp>
#include <iostream>
 
using namespace std;
 
int main(int argc, char **argv) {
  raspicam::RaspiCam_Cv Camera; // Camera Object
  Camera.set(CV_CAP_PROP_FRAME_HEIGHT, 240);
  Camera.set(CV_CAP_PROP_FRAME_WIDTH, 320);
 
  cv::Mat frame;
 
  // Set camera params
  Camera.set(CV_CAP_PROP_FORMAT, CV_8UC3); // For color
 
  // Open camera
  cout << "Opening camera..." << endl;
  if (!Camera.open())
    {
    cerr << "Error opening camera!" << endl; return -1;
    } else
      cout << "OPEN~!!\n";
    //Start capturing
    cv::namedWindow("Display Window", cv::WINDOW_AUTOSIZE);
    int i=0;
    for (;;)
    {
            Camera.set(CV_CAP_PROP_EXPOSURE, 1+i%30);
        Camera.grab();
        Camera.retrieve(frame);
        cv::imshow("Display Window", frame);
        printf("i==%d\n", i);
     if (cv::waitKey(1) > 0)
     {
         break;
     }
     i++;
     if(i<=30)
     {
         cv::imwrite ("image"+std::to_string(i)+".jpg",frame );
     }
    }

  cout << "Stopping camera.." << endl;
  Camera.release();
  return 0;
}
```

By keeping the mapping from 0 to 330000, I just stopped setting the value in 
`  Camera.set(CV_CAP_PROP_EXPOSURE, 1+i%30);`
at 30, therefore I only use range [1, 30] out of [1, 100]. By using values greater than 30, obviously no change in exposure can take place, given the 10 fps.

Below you can see the maximum possible manual exposure with the library compiled having 30 fps vs 10 fps. Camera not moved, light conditions unchanged.

`Camera.set(CV_CAP_PROP_EXPOSURE, 100); // as high as possible shutter speed`
 
[image]

cedricve, 2016-03-15:
As @FranciscMoldovan stated, lowering FPS solved the problem. Images taken with the Raspberry Pi camera module have a better lightning condition. Change has been pushed to develop branch and will be included in the next release with KIOS (2.0.0).

cedricve, 2016-11-29:
Images are still too black, need a way to get brighter images.

Whytehorse, 2016-12-23:
I've worked a lot on this issue. The rpi camera isn't sensitive. The only way to compensate is to change the ISO, FPS, and possibly a couple of other settings. Otherwise you need to add light like an IR illuminator for the NOIR cam or a halogen bulb for the regular cam.

cedricve, 2016-12-24:
@Whytehorse thanks for the share. I agree that if your want a camera which works properly by day and night, you'll need to buy a decent ip camera. Although I'm using a simple USB camera (Logitech C270) with auto focus, which works good at semi-dark conditions.

closed 2016-12-26 by cedricve