kerberos-io/machinery #13: Docker-ize Kerberos-io

JasonSwindle, 2015-07-28:
Howdy,

I can currently working on seeing if I can docker-ize this product. I think huge improvements could be made by shipping the different aspects in containers that link to each other. My step are to get a good build work-flow, which is semi-done. I was able to get a kerberos binary built on my Debian box. What do you think?

cedricve, 2015-07-28:
Hello Jason

Nice. It would be very helpful indeed, I didn't had the time to work on docker. Will you dockerize both machinery and webinterface?

Thanks for your support, we can really use this.

JasonSwindle, 2015-07-28:
# Task List
- [ ] Get a compile-to-docker workflow
- [ ] Make sure `/home/kerberos-io/bin/kerberos` is statically linked
- [ ] deploy docker container to rPi
- [ ] ..... more to come

JasonSwindle, 2015-07-28:
My other idea would be that this product should be like how OpenELEC is setup with buildroot. Buildroot + systemD + Docker would make this a very powerful out of the box system.

cedricve, 2015-07-28:
This sounds robust, and think this would be a huge step forward. Can we open a conversation on gitter?

JasonSwindle, 2015-07-28:
Sure, currently at work, so I may not be able to jump in today. I am also not sure how much I can take on for this project but I can point out what I know. :)

cedricve, 2015-07-28:
No problem, no commitments :)

cedricve, 2015-11-03:
Hello Jason

The kerberosio binary is now completely static compiled (develop branch), do you have any instructions to build the docker workflow?

Regards

JasonSwindle, 2015-11-03:
Howdy,

I will get started on that, and share it today or tomorrow. I had to give
up on my workflow, and things in docker have changed a bit.

Thanks,
Jason

JasonSwindle, 2015-11-08:
I see that ```http://googlemock.googlecode.com/svn/trunk``` has moved to ```https://github.com/google/googlemock```

cedricve, 2015-11-09:
Thanks for the notice, are there any problems with the old repo?

JasonSwindle, 2015-11-09:
I am geting an error that the repo is not found, and google code is going
away soon-ish.... so it maybe time to update your links. :)

cedricve, 2015-11-10:
Ok thanks Jason, I've update the repo url.

cedricve, 2016-01-07:
@JasonSwindle , I've been busy with Docker lately, and I've managed to get a first version of the Dockerfile. It installs both web and machinery repositories. Both services are managed by supervisor.

This is how it looks like for the moment, the only part is to get it working so that a USB camera / Raspberry Pi camera is forwarded to the contaner.

```
FROM phusion/baseimage:0.9.15

MAINTAINER "Cédric Verstraeten" <hello@cedric.ws>

#################################
# Surpress Upstart errors/warning

RUN dpkg-divert --local --rename --add /sbin/initctl
RUN ln -sf /bin/true /sbin/initctl

#############################################
# Let the container know that there is no tty

ENV DEBIAN_FRONTEND noninteractive

#########################################
# Update base image
# Add sources for latest nginx and cmake
# Install software requirements

RUN sudo sed -i -e 's/us.archive.ubuntu.com/archive.ubuntu.com/g' /etc/apt/sources.list
RUN apt-get update && \
apt-get install -y software-properties-common && \
nginx=stable && \
add-apt-repository ppa:nginx/$nginx && \
apt-get update && \
apt-get upgrade -y
RUN apt-get -y install supervisor curl subversion libcurl4-gnutls-dev cmake dh-autoreconf nginx nodejs npm php5-fpm git php5-mysql php-apc php5-curl php5-gd php5-mcrypt php5-memcache php5-tidy php5-xsl pwgen 

########################################
# fix ownership of sock file for php-fpm

RUN sed -i -e "s/;listen.mode = 0660/listen.mode = 0750/g" /etc/php5/fpm/pool.d/www.conf && \
find /etc/php5/cli/conf.d/ -name "*.ini" -exec sed -i -re 's/^(\s*)#(.*)/\1;\2/g' {} \;

###################
# nginx site conf

RUN rm -Rf /etc/nginx/conf.d/* && \
rm -Rf /etc/nginx/sites-available/default && \
mkdir -p /etc/nginx/ssl
ADD ./web.conf /etc/nginx/sites-available/default.conf
RUN ln -s /etc/nginx/sites-available/default.conf /etc/nginx/sites-enabled/default.conf

#####################
# Clone and build web

RUN git clone https://github.com/kerberos-io/web /var/www/web && \
cd /var/www/web && git checkout develop && \
chown -Rf www-data.www-data /var/www/web && \
curl -sS https://getcomposer.org/installer | sudo php -- --install-dir=/usr/local/bin --filename=composer && \
cd /var/www/web && \
php5enmod mcrypt && \
composer install --prefer-source && \
npm install -g bower && \
cd public && \
nodejs /usr/local/bin/bower --allow-root install

############################
# Clone and build machinery

RUN git clone https://github.com/kerberos-io/machinery /tmp/machinery && \
    cd /tmp/machinery && git checkout develop && \
    mkdir build && cd build && \
    cmake .. && make && make check && make install
RUN chown -Rf www-data.www-data /etc/kerberosio
RUN chmod -Rf 777 /etc/kerberosio/config
RUN rm -rf /tmp/machinery

#########################################
# Make capture directory visible and link

RUN rm /var/www/web/public/capture && \
ln -s /srv/capture/ /var/www/web/public/capture

VOLUME ["/srv/capture"]

##############
# Expose Ports

EXPOSE 443
EXPOSE 80

#############################
# Supervisor Config and start

ADD ./supervisord.conf /etc/supervisord.conf
ADD ./run.sh /run.sh
RUN chmod 755 /run.sh

CMD ["/bin/bash", "/run.sh"]
```

I've read a lot of articles which say it's hard to do so. Especially when running docker on (OSX/Windows); you'll need to enable USB filtering on the VM. Do you have some experience with this?

cedricve, 2016-02-24:
docker is added to the different repositories, and a general [docker](https://github.com/kerberos-io/docker/tree/develop) project is created which contains a docker-compose file.

- [docker machinery](https://hub.docker.com/r/kerberos/machinery/)
- [docker web](https://hub.docker.com/r/kerberos/web/)

closed 2016-02-24 by cedricve