kerberos-io/machinery #16: Move S3 upload to machinery and remove sync-s3 repository

cedricve, 2015-11-04:
For performance and architectural reasons, the uploading of images to a S3 bucket should be managed by the machinery and not longer exists as a separate service ([sync-s3](https://github.com/kerberos-io/sync-s3)).

We will need to use the AWS C++ library, and run a seperate thread in the machinery, that will upload images to the S3 bucket. 

Following features are needed
* Upload image to a S3 bucket
* When no connection is available, images needs to be synced at a later time, when the connection is back online.
* Uploading images has a lower priority than the image processing itself, this means that if the machinery is detecting motion and saving images, the sync-thread should halt or release it resources to the image processing.

nicolas-f, 2015-12-27:
If cloud is not configured the process should not be started.

cedricve, 2016-01-28:
Did some research on how to do it in C++: https://blog.cedric.ws/how-to-use-aws-with-your-c-application.
I will soon try to implement this in the machinery.

cedricve, 2016-02-24:
Feature added to develop branch

closed 2016-02-24 by cedricve