kerberos-io/machinery #18: Kerberos stop working, cpu 100%

nicolas-f, 2015-12-26:
Frequently there is no new capture on webcam. The CPU usage is 100% on one core. This is installed on a rasberry pi 2 using your sdcard image. (32 GB)

```top
  PID USER      PR  NI    VIRT    RES  %CPU %MEM     TIME+ S COMMAND
29528 root      20   0  125.5m  47.3m 120.9  5.5   0:06.11 R grunt
29478 pi        20   0    7.0m   2.9m   2.0  0.3   0:01.01 R top
    7 root      20   0    0.0m   0.0m   0.7  0.0  11:08.51 S rcu_preempt
  139 root      20   0   80.4m  42.9m   0.7  5.0  12:19.35 S systemd-journal
    1 root      20   0   23.3m   4.9m   0.0  0.6   3:26.24 S systemd
    2 root      20   0    0.0m   0.0m   0.0  0.0   0:00.07 S kthreadd
    3 root      20   0    0.0m   0.0m   0.0  0.0   0:49.81 S ksoftirqd/0
    5 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kworker/0:0H
    8 root      20   0    0.0m   0.0m   0.0  0.0   0:00.05 S rcu_sched
    9 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S rcu_bh
   10 root      rt   0    0.0m   0.0m   0.0  0.0   0:00.81 S migration/0
   11 root      rt   0    0.0m   0.0m   0.0  0.0   0:00.72 S migration/1
   12 root      20   0    0.0m   0.0m   0.0  0.0   0:15.79 S ksoftirqd/1
   14 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kworker/1:0H
   15 root      rt   0    0.0m   0.0m   0.0  0.0   0:00.46 S migration/2
   16 root      20   0    0.0m   0.0m   0.0  0.0   0:14.74 S ksoftirqd/2
   18 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kworker/2:0H
   19 root      rt   0    0.0m   0.0m   0.0  0.0   0:00.51 S migration/3
   20 root      20   0    0.0m   0.0m   0.0  0.0   0:11.60 S ksoftirqd/3
   22 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kworker/3:0H
   23 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S khelper
   24 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S kdevtmpfs
   25 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S netns
   27 root      20   0    0.0m   0.0m   0.0  0.0   0:00.19 S khungtaskd
   28 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S writeback
   29 root      25   5    0.0m   0.0m   0.0  0.0   0:00.00 S ksmd
   30 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S crypto
   31 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kintegrityd
   32 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S bioset
   33 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kblockd
   34 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S md
   36 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S rpciod
   52 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S kswapd0
   53 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S fsnotify_mark
   54 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S nfsiod
   55 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S jfsIO
   56 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S jfsCommit
   57 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S jfsCommit
   58 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S jfsCommit
   59 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S jfsCommit
   60 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S jfsSync
   61 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S xfsalloc
   62 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S xfs_mru_cache
   69 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S pencrypt
   70 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S pdecrypt
   77 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kthrotld
   78 root       1 -19    0.0m   0.0m   0.0  0.0   0:00.00 S VCHIQ-0
   79 root       1 -19    0.0m   0.0m   0.0  0.0   0:00.00 S VCHIQr-0
   80 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S VCHIQs-0
```

All is resolved when I restart. But the problem come back again in 1-2 days.

Thank you

cedricve, 2015-12-26:
Hello Nicolas

We are aware of this problem, it occurs when there are a lot of images in the image directory; e.g. expanded SD card in your case 32GB. The grunt process (a file watcher) running in the background is used to sync the images to our cloud application. Grunt uses a lot of CPU when the amount of images stored in the directory is huge. We're optimising the sync function, and will integrate it directly in the machinery in a next release #16. 

If you are not using our cloud application, and are not planning to use it you can simple disable it from the system (https://doc.kerberos.io/1.0/addons/Cloud). 

Just run following command, after signing in with ssh, and reboot the system:

`
systemctl disable cloud.service
`

Hope this helps..

nicolas-f, 2015-12-27:
Ok thank you. I did't recognize my issue in this ticket.

nicolas-f, 2015-12-27:
The folder is 767 MB

nicolas-f, 2015-12-28:
cpu usage is now ok. But kerberos still cut itself after a while. I don't know the log file you need.

```
[pi@alarmpi logs]$ top
top - 09:36:52 up 15:18,  1 user,  load average: 0.00, 0.01, 0.05
Tasks: 103 total,   1 running, 102 sleeping,   0 stopped,   0 zombie
%Cpu0  :   0.0/0.0     0[                                                     ]
%Cpu1  :   1.3/1.9     3[||                                                   ]
%Cpu2  :   0.0/0.0     0[                                                     ]
%Cpu3  :   0.0/0.0     0[                                                     ]
GiB Mem :  6.3/0.839    [                                                     ]
GiB Swap:  0.0/0.000    [                                                     ]

  PID USER      PR  NI    VIRT    RES  %CPU %MEM     TIME+ S COMMAND
  139 root      20   0  152.9m  28.8m   0.0  3.3   0:06.59 S systemd-journal
  241 root      20   0   62.4m  12.2m   0.0  1.4   0:05.06 S php-fpm
  246 http      20   0   62.6m  11.2m   0.0  1.3   0:18.96 S php-fpm
  247 http      20   0   62.6m  10.9m   0.0  1.3   0:17.14 S php-fpm
  249 http      20   0   62.6m  10.9m   0.0  1.3   0:16.79 S php-fpm
    1 root      20   0   24.3m   4.9m   0.0  0.6   0:51.34 S systemd
  215 root      20   0    7.6m   4.3m   0.0  0.5   0:11.04 S haveged
 5741 root      20   0    9.3m   4.2m   0.0  0.5   0:00.25 S sshd
  242 root      20   0    6.8m   3.9m   0.0  0.5   0:00.08 S sshd
 5743 pi        20   0    9.4m   3.3m   0.0  0.4   0:02.03 S sshd
  219 systemd+  20   0   12.1m   3.2m   0.0  0.4   0:00.89 S systemd-network
  154 root      20   0   11.5m   3.1m   0.0  0.4   0:00.84 S systemd-udevd
  217 dbus      20   0    6.3m   3.1m   0.0  0.4   0:08.32 S dbus-daemon
  208 systemd+  20   0   16.0m   3.0m   0.0  0.3   0:00.54 S systemd-timesyn
  245 http      20   0    6.2m   3.0m   0.0  0.3   0:00.89 S nginx
 5870 pi        20   0    7.0m   2.9m   2.6  0.3   0:03.66 R top
 5858 root      20   0    6.1m   2.8m   0.0  0.3   0:00.02 S crond
  243 systemd+  20   0    4.1m   2.8m   0.0  0.3   0:00.77 S systemd-resolve
 5744 pi        20   0    3.5m   2.7m   0.0  0.3   0:00.37 S bash
  216 root      20   0    3.5m   2.6m   0.0  0.3   0:04.47 S systemd-logind
 5859 root      20   0    3.2m   2.0m   0.0  0.2   0:00.02 S sh
  212 root      20   0    3.1m   1.9m   0.0  0.2   0:00.53 S crond
  244 root      20   0    5.9m   1.6m   0.0  0.2   0:00.00 S nginx
  222 root      20   0    2.0m   1.5m   0.0  0.2   0:00.02 S agetty
  221 root      20   0    2.0m   1.4m   0.0  0.2   0:00.01 S agetty
 5869 root      20   0    1.9m   0.4m   0.0  0.0   0:00.00 S sleep
    2 root      20   0    0.0m   0.0m   0.0  0.0   0:00.05 S kthreadd
    3 root      20   0    0.0m   0.0m   0.0  0.0   0:05.77 S ksoftirqd/0
    5 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kworker/0:0H
    7 root      20   0    0.0m   0.0m   0.0  0.0   0:20.76 S rcu_preempt
    8 root      20   0    0.0m   0.0m   0.0  0.0   0:00.01 S rcu_sched
    9 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S rcu_bh
   10 root      rt   0    0.0m   0.0m   0.0  0.0   0:00.05 S migration/0
   11 root      rt   0    0.0m   0.0m   0.0  0.0   0:00.07 S migration/1
   12 root      20   0    0.0m   0.0m   0.0  0.0   0:00.05 S ksoftirqd/1
   14 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kworker/1:0H
   15 root      rt   0    0.0m   0.0m   0.0  0.0   0:00.06 S migration/2
   16 root      20   0    0.0m   0.0m   0.0  0.0   0:00.03 S ksoftirqd/2
   18 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kworker/2:0H
   19 root      rt   0    0.0m   0.0m   0.0  0.0   0:00.06 S migration/3
   20 root      20   0    0.0m   0.0m   0.0  0.0   0:00.02 S ksoftirqd/3
   22 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kworker/3:0H
   23 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S khelper
   24 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S kdevtmpfs
   25 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S netns
   26 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S perf
   27 root      20   0    0.0m   0.0m   0.0  0.0   0:00.05 S khungtaskd
   28 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S writeback
   29 root      25   5    0.0m   0.0m   0.0  0.0   0:00.00 S ksmd
   30 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S crypto
   31 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kintegrityd
   32 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S bioset
   33 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S kblockd
   34 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S md
   36 root       0 -20    0.0m   0.0m   0.0  0.0   0:00.00 S rpciod
   52 root      20   0    0.0m   0.0m   0.0  0.0   0:00.00 S kswapd0
```

nicolas-f, 2015-12-28:
I run the command (systemctl status):

```
[pi@alarmpi system]$ sudo systemctl status kerberos.service
* kerberos.service - Kerberos.io Video Surveillance
   Loaded: loaded (/etc/systemd/system/kerberos.service; enabled; vendor preset: disabled)
   Active: failed (Result: core-dump) since Mon 2015-12-28 06:14:07 UTC; 3h 28min ago
 Main PID: 214 (code=dumped, signal=SEGV)

Dec 27 18:18:22 alarmpi systemd[1]: Starting Kerberos.io Video Surveillance...
Dec 27 18:18:25 alarmpi kerberos[214]: OpenCV Error: Unspecified error (GStreamer: your gstreamer installation is missing a required plugin
Dec 27 18:18:25 alarmpi kerberos[214]: ) in handlemessage, file /home/kerberos-io/build/opencv/modules/highgui/src/cap_gstreamer.cpp, line 1622
Dec 28 06:13:56 alarmpi kerberos[214]: OpenCV Error: Unspecified error (GStreamer: your gstreamer installation is missing a required plugin
Dec 28 06:13:56 alarmpi kerberos[214]: ) in handlemessage, file /home/kerberos-io/build/opencv/modules/highgui/src/cap_gstreamer.cpp, line 1622
Dec 28 06:14:07 alarmpi systemd[1]: kerberos.service: Main process exited, code=dumped, status=11/SEGV
Dec 28 06:14:07 alarmpi systemd[1]: Failed to start Kerberos.io Video Surveillance.
Dec 28 06:14:07 alarmpi systemd[1]: kerberos.service: Unit entered failed state.
Dec 28 06:14:07 alarmpi systemd[1]: kerberos.service: Failed with result 'core-dump'.
Dec 28 06:14:08 alarmpi systemd-coredump[4649]: Process 214 (kerberos) of user 0 dumped core.
```

cedricve, 2015-12-28:
Please join our gitter channel, to discuss this issues: https://gitter.im/kerberos-io/hades

securitybyte, 2015-12-31:
Hi 

My Pi stopped capturing sometime over the night too. 

The CPU was *not* at 100%, and there was nothing in the log.stash. I did not notice if the kerberos.service was running..

I had already disabled the cloud service yesterday, and my capture directory only has about 100 images totalling 4 megs.

I didn't want to raise as a new issue without doing some troubleshooting first. Is there anything I should check if it happens again? (Apart from systemctl status kerberos.service ?)

Many thanks

securitybyte, 2016-01-01:
Hi, looks like 

```
sudo systemctl status kerberos.service -l
* kerberos.service - Kerberos.io Video Surveillance
   Loaded: loaded (/etc/systemd/system/kerberos.service; enabled; vendor preset: disabled)
   Active: failed (Result: core-dump) since Thu 2015-12-31 17:41:49 UTC; 18h ago
  Process: 201 ExecStart=/home/kerberos-io/bin/kerberos (code=dumped, signal=SEGV)                                                               Main PID: 201 (code=dumped, signal=SEGV)

Dec 31 15:00:03 alarmpi systemd[1]: Starting Kerberos.io Video Surveillance...
Dec 31 15:00:05 alarmpi kerberos[201]: OpenCV Error: Unspecified error (GStreamer: your gstreamer installation is missing a required plugin
Dec 31 15:00:05 alarmpi kerberos[201]: ) in handlemessage, file /home/kerberos-io/build/opencv/modules/highgui/src/cap_gstreamer.cpp, line 1622
Dec 31 17:41:39 alarmpi kerberos[201]: OpenCV Error: Unspecified error (GStreamer: your gstreamer installation is missing a required plugin
Dec 31 17:41:39 alarmpi kerberos[201]: ) in handlemessage, file /home/kerberos-io/build/opencv/modules/highgui/src/cap_gstreamer.cpp, line 1622
Dec 31 17:41:49 alarmpi systemd[1]: kerberos.service: Main process exited, code=dumped, status=11/SEGV                                          Dec 31 17:41:49 alarmpi systemd[1]: Failed to start Kerberos.io Video Surveillance.                                                             Dec 31 17:41:49 alarmpi systemd[1]: kerberos.service: Unit entered failed state.                                                                Dec 31 17:41:49 alarmpi systemd[1]: kerberos.service: Failed with result 'core-dump'.                                                           Dec 31 17:41:50 alarmpi systemd-coredump[1261]: Process 201 (kerberos) of user 0 dumped core.                                                                                                   Stack trace of thread 201:                                                                                                                      #0  0x0000000000000020 n/a (n/a)                  

```

cedricve, 2016-01-03:
For the moment it's unclear why opencv and gstreamer fail. We'll investigate this problem further.
Thank you both for your support so far!

cedricve, 2016-02-24:
Solved on develop branch

closed 2016-02-24 by cedricve

nicolas-f, 2016-02-25:
Hi,

Thank you. How can we update a kerberos image sdcard (running version) with the develop branch version ?

Thank you

cedricve, 2016-02-25:
We'll release a new image soon Nicolas, keep you posted ;)

nicolas-f, 2016-02-25:
Thanks.
Can I upgrade Kerberos without having to reformat the sd card ?

cedricve, 2016-02-25:
Yes it but will give you some manual work unfortunately (you need to recompile the develop branch). We are working on a new project called KIOS. This will create a more lightweight Linux OS.