kerberos-io/machinery #19: Mjpeg Stream -> 'GStreamer - missing plugin' Error

securitybyte, 2015-12-31:
Hi there

I am running Kerberos v1.1.0 on a Pi2, installed via the kerberos-io-armv7-4GB-v110 image. I am using it with a D-Link DCS-932LB IP camera.

If I configure Kerberos to use the mjpeg stream (which I believe is from http://user:pass@x.x.x.x/video.cgi), then it does not capture anything. I confirm this URL does work in the browser.

The log.stash gets the following error written to it every 10 seconds:
```
31-12-2015 10:10:26 4 - OpenCV : /home/kerberos-io/build/opencv/modules/highgui/src/cap_gstreamer.cpp:1622: error: (-2) GStreamer: your gstreamer installation is missing a required plugin
 in function handlemessage
```

Alternatively, I can use the http://user:pass@x.x.x.x/image.jpg instead, which works fine. I am not sure whether there is any benefit in my case of using the mjpeg stream over the image.jpg, but thought I'd let you know. 

Please let me know if I can provide any more details.

Many thanks

cedricve, 2016-01-03:
IP camera with mjpeg stream fails at random intervals #18

cedricve, 2016-02-24:
Solved on develop branch

closed 2016-02-24 by cedricve