kerberos-io/machinery #21: enable RTSP stream for IP Cameras

mazki516, 2016-01-21:
instead of connecting to the mjpeg url of the ip camera ,
we will connect to the rtsp url which is more suitable for stream and have more control over the camera.

cedricve, 2016-02-24:
Feature added to develop branch

closed 2016-02-24 by cedricve

mcgyver83, 2017-01-04:
Hi all, I'm running kerberos.io on a raspberry pi2 but I cannog get my Sricam ONVIF ipcam working with RTSP streaming.
This is working with vlc and rtsp://192.168.0.101:554/onvif1 but I cannot get it working with kerberos :(

```
04/01/2017 22:03:48.714 INFO  [trivial] Starting capture device: McG-camera
04/01/2017 22:03:48.715 ERROR [trivial] Kerberos : Factory : Could not create an instance of message: McG-camera
```

```
<streams>
    <Rtsp>
        <enabled type="bool">true</enabled>
        <streamPort type="number">554</streamPort>
        <quality type="number">75</quality>
    </Rtsp>
</streams>

<captures>
    <IPCamera>
        <url type="text">rtsp://192.168.0.101:554/onvif1</url>
        <frameWidth type="number">640</frameWidth>
        <frameHeight type="number">360</frameHeight>
        <delay type="number">1000</delay>
        <angle type="number">0</angle>
    </IPCamera>
</captures>
```