kerberos-io/machinery #24: Execute multiple kerberos instance on a machine

cedricve, 2016-01-28:
It would be helpfull, if multiple camera's can be processed on a single machine. Therefore it should be possible to run one or more instances.

To make this possible a user should be able to pass parameters which can override the exisiting configuration files.

E.g.

```
kerberosio --capture IPCamera --captures.IPCamera.url rtsp://camera1--name ipcamera1

kerberosio --capture IPCamera --captures.IPCamera.url rtsp://camera2--name ipcamera2

kerberosio --capture IPCamera --captures.IPCamera.url rtsp://camera3--name ipcamera3
```

cedricve, 2016-01-28:
fix added to develop branch 

https://github.com/kerberos-io/machinery/commit/968db17e3e4d7a7375155c59fb187d5b655f4c64

closed 2016-01-28 by cedricve

marksev1, 2016-12-07:
I don't understand something from the docs...If I would have multiple cameras set up, could I still view them all in the WebUI and see events of all of them there?

cedricve, 2016-12-07:
Yes, that's correct. All images will be shown in the web interface.

marksev1, 2016-12-07:
In the same web interface? Or would one have to know the specific IP address of each camera, to invoke each different webui?
If its not in the same interface can you make tabs in the webui to switch between various cameras?

cedricve, 2016-12-07:
This topic is about multiple camera's on the same machine. Your question is different: can I view images of multiple machines in one web interface. 

We have created a cloud application to solve this problem (https://cloud.kerberos.io). The idea is that you subscribe to a plan, and configure every machines with some credentials, and from then they will sync images to our cloud application.

muten84, 2016-12-07:
Each machinery instance can manage 1 camera and the web ui can read an image folder and can manage config files of a single instance.