kerberos-io/machinery #25: New repository: Pilot

cedricve, 2016-02-12:
A new repository will be created, named Pilot (other names are welcome). The Pilot project will provide a GUI in the terminal, to start and monitor multiple kerberos instances (multiple machinery running). This will allow us to control one or more camera's on a single Raspberry Pi.

- [ ]  Create a first mockup of the GUI.
- [ ]  Implement ONVIF standard to locate IP camera's available on the network.
- [ ]  Define one or more IP camera's, USB camera or Rapsberry Pi Camera (and specify additional configuration, e.g. RTSP url).  
- [ ]  Run a machinery process for each camera with there specific configuration.
- [ ]  Show a real-time view of all the machinery processes running (memory consuming, etc)
- [ ]  more to come ..

closed 2016-06-07 by cedricve

Orabig, 2016-06-08:
This idea has been cancelled ?

cedricve, 2016-06-08:
I don't have the time right now, and @mazki516 is working for a client. I think I will redesign the machinery and web to handle multiple instances. Future work..