kerberos-io/machinery #30: Buildroot set-up to build for multiple boards

cedricve, 2016-02-20:


ccrisan, 2016-02-24:
I have prepared a stripped-down version of motionEyeOS (basically just the OS w/o motion or motionEye). It's not a fork *per se* and I don't think it should be. It's currently on my computer waiting to be pushed somewhere.

I took the liberty to name the OS "kios" as I needed a name. Please feel free to suggest any other name.

I believe a repo should be created for it and we should start "moving" Kerberios.io-specific packages to the distro's build system, so that everything gets built automatically with one command (I'd be happy to help adding these custom packages to the BuildRoot system).

cedricve, 2016-02-24:
Great job ccrissan! I've created a repo called "kios", which contains 2 branches "master" and "develop".
We can push a first version to the develop [branch](https://github.com/kerberos-io/kios/tree/develop).

closed 2016-02-24 by cedricve