kerberos-io/machinery #31: IO - run arbitrary script/program

discoduck2x, 2016-02-23:
I want to hook onto gammu to send txt messages , now i cant see how to do that currently?
(unless i just trigger by image creation outside of kerberos)

cedricve, 2016-02-23:
For the moment not possible, we'll need to add a new IO device, which gets as parameter the path to a script.

marksev1, 2016-12-07:
Events are sent via Webhooks right? Maybe you could tie that with your service for sending msg's discoduck2x?

discoduck2x, 2016-12-07:
@marksev1 that means one more thing inbetween , need something to listen for the weebhook post and then have that trigger script/gammu sms process..... just let it run any arbitrary script with stuff as variables imgs etc....

cedricve, 2017-01-31:
feature has been added to develop branch, and will be integrated in version 2.2.0

closed 2017-02-13 by cedricve