kerberos-io/machinery #32: Bloated behaviour

discoduck2x, 2016-02-23:
Why is it calling the hades.kerberos website by every run/clean script and why is it accessing bitbucket every 2 seconds?

cedricve, 2016-02-23:
Hello discoduck.

Please show me where it's accessing bitbucket, this isn't build in.
The reason for polling the webpage is to keep the WIFI connection alive, we have found situation that the WIFI connection went to sleep and no longer was accessible.

discoduck2x, 2016-02-23:
hello,

i saw it in my firewall , every 2 seconds its making a http call to 169.254.169.254 (which according to ripe is bitbucket) ,, ,i checked a netstat -natp and its the grunt process that is doing it. didnt check further (im using your SD image btw so its nothing i have installed..)

regarding the wifi that is solved by the fix you´ve already seem to have done with no powersaving and the special line in /etc/modprobe.d/8192cu.conf  , , atleast thats good enough to not have the wifi power down during inactivity.

discoduck2x, 2016-02-23:
@cedricve above

cedricve, 2016-02-23:
Hello disco

Good point.. I'll we remove this from next release.

Regarding the bitbucket call, this is strange, we need to look at this more in detail? It's not anything I've installed on top Arch Linux.

ccrisan, 2016-02-23:
Isn't 169.254.169.254 an auto configured IP? What does it have to do with Bitbucket? Or did I misread something?

discoduck2x, 2016-02-23:
@ccrisan lol you are right , i was too fast on the trigger,, just saw it in my firewall sourced from the Pi on our guest wifi network so i pasted it on ripe and looked and saw bitbucket@ripe.net ...seems i didnt care to read the first lines that says private  ...my mistake i appologise. 

still ,, what is grunt doing? i dont have my image running anymore so i cant check again for contents of packets,, but just do a netstat and im sure ull see it

cedricve, 2016-02-23:
Grunt is a filewatcher which pushed images (secured) to the cloud application (AWS). In a next release this will be move completely in the machinery itself. #16

discoduck2x, 2016-02-23:
sorry again for wasting your time @cedricve & @ccrisan

closed 2016-02-23 by discoduck2x