kerberos-io/machinery #33: mjpeg error from ipcamera

muten84, 2016-03-14:
Hi i can't get any image from my ipcamera setted in kerberos.io configuration, i tried the url configured in kerberos-web from the browser, and it works fine. When i start kerberos the log of the machinery says:
OpenCV : /home/pi/kerberos-io/build/opencv/modules/imgproc/src/color.cpp:3739: error: (-215) scn == 3 || scn == 4 in function cvtColor
I can't understand if depends on my ipcamera stream because i don't have another ipcamera to test.
The disk output is set correctly i can't understand what's the problem. How can i test a MJPG stream to contribute debugging? 

Thanks in advance.

Luigi B.

cedricve, 2016-03-15:
This is a know issue in the current release. The problem will be resolved in a next release 2.0.0. E.A.T. is end of April. #19 #18

muten84, 2016-03-16:
i recompiled by myself the develop branch and it seems that now ipcam url was never invoked (i can check this with an http proxy).

cedricve, 2016-03-17:
Do you have the possibility to check with an RTSP connection? Are you sure ffmpeg is installed?

muten84, 2016-03-20:
Hi sorry for late response. I dont have RTSP option, i tried to convert  the mjpeg stream trough a RTSP server but i can't get a valid stream, neither this try can be helpful. However i figured out that VideoCapture class of opencv library does not work if the url is not extended with a ".mjpg" , when you want to capture mjpeg streams. So i've tried switching again to latest stable release and changing the url in the settings page of kerberos-web (adding "&dummy=param.mjpg" it's enough), and now i see capturing screenshots correctly and the error of the cvtColor function is logged very rarely. My ipcam has IR Led for night vision, i'm going to try the behavior correctness during night also. Hoping these info will be useful. 
For the develop branch all i can say is that the capture in my opinion is not starting, another doubt i have is if I must use latest  kerberos-web with the latest kerberos-io or is not relevant. Is there a way to debug the kerberos-io activity enabling a debug verbosity or something similar?

Thanks for all.

Luigi

cedricve, 2016-03-21:
Nice great research, well done! If you are using the develop branch of the machinery, I recommend you should also use the develop branch of the web cient. Give it a try..

closed 2016-04-20 by cedricve