kerberos-io/machinery #34: raspberry pi 3

sirbarret, 2016-03-17:
Hi

I cant get the SD Image to run in pi 3

cedricve, 2016-03-17:
RPi3 isn't supported for the moment. We are working on it.

closed 2016-03-17 by cedricve

cedricve, 2016-04-05:
RPi3 support will be added in next release.

mgreau, 2016-08-19:
Hi, 
I installed KiOS on my RPi3 but the camera doesn't work.

I have only black images and no live view.
If I connect to the RPi3 with ssh, I have the following result:

----
[root@kios-2ab19586 ~]# raspistill -v

raspistill Camera App v1.3.8

Width 2592, Height 1944, quality 85, filename (null)
Time delay 5000, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates
----

On the website, it is noticed that the RPi3 is supported https://doc.kerberos.io/2.0/requirements
But I don't understand the status of this issue.

It seems to be the problem described here: http://raspberrypi.stackexchange.com/questions/13764/what-causes-enospc-error-when-using-the-raspberry-pi-camera-module

Do you have a workaround?

Thanks and nice job by the way :)

cedricve, 2016-08-19:
hello @mgreau thanks for the report. Are you using the latest RPi camera v2 ?

mgreau, 2016-08-19:
I use the RPi Camera v2.1

cedricve, 2016-08-19:
We are preparing a new release with some new features, and also support for the v2.1 :)
Can you try this beta release to check it out? I got the v2.1 next week so I couldn't test it myself.

https://github.com/cedricve/kios/releases/tag/develop

mgreau, 2016-08-19:
Yes sure, I will try it now, thanks

mgreau, 2016-08-19:
do I have to format the SD card before installing the new image?
I follow the same steps as the first installation (OSX) but the KiOS doesn't seem to be updated.

cedricve, 2016-08-19:
Normally, that wouldn't be necessary. However you can try it to be sure.

mgreau, 2016-08-19:
the new KiOS version is installed:

[root@kios-2ab19586 ~]# uname -a
Linux kios-2ab19586 4.4.12-v7 #1 SMP Fri Aug 12 01:53:35 EDT 2016 armv7l GNU/Linux

But unfortunately I have the same problem in the HTML UI and with the `raspistill -v` command:

ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

cedricve, 2016-08-19:
Hmm ok, that's bad news. I'll check it.

mgreau, 2016-08-19:
How can I update the `/boot/cmdline.txt` file to test some configuration proposed here? http://raspberrypi.stackexchange.com/questions/13764/what-causes-enospc-error-when-using-the-raspberry-pi-camera-module

I can't save the file with vi because of `'/boot/cmdline.txt' is read only` and `chmod: /boot/cmdline.txt: Read-only file system`

cedricve, 2016-08-19:
Good point :)

Edit the file on the data partition, and change debug to true. You could also plug-it back into your host OS and change it.

nano /data/etc/os.conf

cedricve, 2016-08-19:
Hmm apparently I forgot to update the firmware package of userland. I'm currently recompiling everything so that latest changes are included. Hold on :)

mgreau, 2016-08-19:
Cool, because my camera is connected 
`$ vcgencmd get_camera
supported=1 detected=1 `

but I still have the problem even after editing the `config.txt` file and reboot after adding the following line:
` gpu_mem_1024=944`

* doc: https://www.raspberrypi.org/documentation/configuration/config-txt.md
* forum: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=150994&p=1022544&hilit=ENOSPC#p1022544

cedricve, 2016-08-19:
indeed @mgreau, I've just uploaded some new images can you give it a second try?

https://github.com/cedricve/kios/releases/tag/develop

mgreau, 2016-08-20:
Yeah it works 😀
Thanks! 👍