kerberos-io/machinery #36: Add timestamp to the images

Orabig, 2016-04-30:


cedricve, 2016-06-06:
This should be an option in the configuration files, which can be enabled/disabled. @Orabig is there a preferred location for the timestamp? Left-top or right-bottom?

Orabig, 2016-06-06:
Well, I don't think it matters a lot.
I just made some research, and it seems the timestamps can be in any of the corners.

cedricve, 2016-06-07:
User can now check an option if he want to mark the images with a timestamp (timezone taken into account) + the color of the timestamp. Location is always left top. If needed we can make this configurable as well in the future.

[image]

commit 75c4182f (2016-06-07): add possibility to mark images with timestamp (user can specify color), this closes #36
closed 2016-06-07 by cedricve