kerberos-io/machinery #37: Expositor: multiple regions

muten84, 2016-05-01:
It would be useful to have an expositor for configuring more than one region. The behavior could be like the Hull Expositor to paint N rectangles on the image. This kind of expositor should help to monitor many sub-regions, for outdoor surveillance context and to avoid monitoring of non sensible areas, such as pedestrians passage.

cedricve, 2016-06-07:
I've been thinking about this and basically you can fake it with one simple expositor. See my image blow. I don't see a huge benefit in function of the development effort that is required to create this.

What do you think @muten84 ?

[image]

muten84, 2016-06-07:
Great! It should be ok. Let me try this approach with my scenarios so I can check it.

muten84, 2016-06-09:
Tried on latest stable version: it works!

closed 2016-06-09 by muten84