kerberos-io/machinery #38: Enabling machinary fails

aladagemre, 2016-05-01:
I followed the installation tutorial for Raspian on Raspberry Pi 1. I had to run "sudo apt-get install -y systemd systemd-sysv" to get systemctl. It even didn't exist beforehand. But failed at this step:

> pi@raspberrypi ~ $ sudo systemctl enable kerberosio
> Failed to issue method call: Invalid argument

What could be the cause?

cedricve, 2016-05-02:
That's weird systectl should be available out-of-the-box. Which version of Raspbian are you running?

aladagemre, 2016-05-03:
/etc/debian_version says 7.10

aladagemre, 2016-05-03:
I tried to dist-upgrade but failed since it could not find locale command. This is due to corrupted filesystem or SD Card I guess. All of these could be due to being unable to install some of the packages either during OS install or afterwards. My case is not sterile enough, so closing the issue.

closed 2016-05-03 by aladagemre