kerberos-io/machinery #41: More instances streaming port

muten84, 2016-05-18:
When runnning more than one instance on same machine i can't find any way to change the streaming port of the second instance. Is that feature enabled? Can i keep running second instance despite the bind sock error?

cedricve, 2016-05-19:
[Port is fixed](https://github.com/kerberos-io/machinery/blob/master/src/kerberos/Kerberos.cpp#L279) in the source code for the moment. I think we'll need to move this setting to the config files; despite that the webinterface only supports one stream. 

However you would not be able to run kerberos with the bind errors, I'll fix this in the next release.

muten84, 2016-05-20:
Hi thanks for all informations. I really need this feature so i made a patch by myself, passing  a StringMap parameter to the startStreamThread method too. After a recompile it's working well, i don't know if you imagined to do same thing in the code. I modified Kerberos.h and Kerberos.cpp files and i used a convenient config param "captures.streamPort". All changes i've done was made locally i didn't fork anything. Waiting for next release.... did you expect to develop a "switch instance" functionality on the web client? Thanks in advance for all.

Luigi

cedricve, 2016-05-20:
Hi Luigi

Great, I was planning to do the change myself, but If you would be able to fork the repository and push the changes, that would be great. I can integrate the possibility of multiplestreams in the web interface, maybe we can discuss this point?

Cédric

muten84, 2016-05-20:
My two cents :) #43

closed 2016-05-20 by cedricve