kerberos-io/machinery #45: Raspberry Pi 2.1 camera doesn't work

cedricve, 2016-06-06:


cedricve, 2016-06-07:
Someone reported that he's able to use the Raspberry Pi v2.1 with Kerberos.io. The only problem is that it doesn't work with specific resolutions. 

3280 × 2464 black image in interface and in x.x.x.x:8888, not working... 
1640 × 1232 scrambled image in interface and in x.x.x.x:8888, not 
working... see image joint 
1408 × 792 ok 
1280 × 720 ok 
960 × 540 ok

To be confirmed, can someone test this out?

muten84, 2016-06-07:
Tomorrow i will have to buy a 4MP IPCamera, do you think it's same test condition? i think it could be possible to switch resolution trough IPCamera web console provided by third-party vendor. 
If so, as soon as possible I can check it. Stay tuned for updates.

Luigi

cedricve, 2016-06-07:
Hmm, it only appears to be a problem for the newest Raspberry Pi Camera module. Myself used some IP-camera's so far (4K as well) and it worked properly with Kerberos.io. I think it's happening under the hood with the Raspicam library which is been used when selecting the Raspi capture device.

closed 2016-08-20 by cedricve