kerberos-io/machinery #50: service hangs after few hours

NicoVermeir, 2016-08-25:
I'm not sure if this is the correct repo to report this issue.

I installed KerberosIO on Raspbian (rpi2), it works fine with a usb webcam, but after a few hours the feed is not available anymore and I need to restart the service. The log doesn't tell me anything either, last entry is "Starting stream on port 8888"

cedricve, 2016-08-25:
Hello Nico

Did you enable verbose logging already?

NicoVermeir, 2016-08-25:
Hi Cedric,

just switched it on, I'll keep an eye on it. thanks

NicoVermeir, 2016-08-25:
well, it happened again. this is in the logs:
[image]

extra info: I have not changed anything in the config, only switched to usb camera and that was it

cedricve, 2016-08-25:
Thank you Nico, can you attach the config.xml and condition.xml

NicoVermeir, 2016-08-25:
here you go
[configs.zip](https://github.com/kerberos-io/machinery/files/437163/configs.zip)

cedricve, 2016-08-25:
Thanks, is there a reason why you've changed "Enabled" to false in the condition.xml?

NicoVermeir, 2016-08-25:
I haven't changed anything in the condition settings AFAIK

cedricve, 2016-08-25:
Hmmm, I will check this. Can you change it back to true, and see if the problem still occurs?

NicoVermeir, 2016-08-25:
will do!

NicoVermeir, 2016-08-26:
the issue is still there, but this time the log looks like this:
[image]

cedricve, 2016-08-26:
Hmm, interesting. Please attach the complete log file.

NicoVermeir, 2016-08-26:
[log.zip](https://github.com/kerberos-io/machinery/files/438623/log.zip)

cedricve, 2016-09-20:
@NicoVermeir can you try with latest version?

NicoVermeir, 2016-09-28:
seems to be resolved. thanks!

closed 2016-09-28 by NicoVermeir