kerberos-io/machinery #51: build fail in raspbian jessie

ypwu1, 2016-08-30:
Build failed in installation step as following:

`pi@raspberrypi:~/machinery/build $ cmake .. && make && make check && sudo make install
External project: Googletest
External project: OpenCV
Bind project: Userland
External project: RaspiCam
External project: Restclient
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/machinery/build
[  8%] Built target googletest
[ 16%] Built target opencv
[ 24%] Built target raspicamera
[ 32%] Built target restclient
[ 36%] Built target ENCODE_LIBRARY
[ 36%] Built target rerun
[ 41%] Built target FILEWATCHER_LIBRARY
[ 42%] Built target EXECUTOR_LIBRARY
[ 46%] Built target TINYXML_LIBRARY
[ 60%] Built target KERBEROS_CORE
[ 61%] Built target MMAPGPIO_LIBRARY
[ 78%] Built target kerberosio
Linking CXX executable ../../test/bin/kerberos-test
/usr/bin/ld: cannot open output file ../../test/bin/kerberos-test: No such file or directory
collect2: error: ld returned 1 exit status
test/CMakeFiles/kerberos-test.dir/build.make:607: recipe for target '../test/bin/kerberos-test' failed
make[2]: *** [../test/bin/kerberos-test] Error 1
CMakeFiles/Makefile2:1536: recipe for target 'test/CMakeFiles/kerberos-test.dir/all' failed
make[1]: *** [test/CMakeFiles/kerberos-test.dir/all] Error 2
Makefile:147: recipe for target 'all' failed
make: *** [all] Error 2`

cedricve, 2016-08-30:
can you try sudo make?

cedricve, 2016-08-30:
can you try 

> sudo make

ypwu1, 2016-08-30:
Fantastic! issue solved.

closed 2016-08-30 by ypwu1