kerberos-io/machinery #54: Webhook is not working with slack

ypwu1, 2016-09-08:
I created a slack channel and subscribe channel for webhook, I tested  with postman which works fine but not working on machinery.

slack webhook is like: `https://hooks.slack.com/services/<your-webhook-api-key>`

cedricve, 2016-09-09:
Thanks for the report, I'll try this myself. Keep you posted @ypwu1 .

cedricve, 2016-09-09:
A sneek peak shows that Slack expects a json object with a text property; https://api.slack.com/incoming-webhooks. Unfortunetaly the format of the JSON object, which is posted by the IO webhook is slightly different.

    [
        'regionCoordinates' => array (
            0 => 677,
            1 => 302,
            2 => 782,
            3 => 534,
        ),
        'numberOfChanges' => 2396,
        'timestamp' => '1473428717',
        'microseconds' => '6-577591',
        'token' => 869,
        'pathToImage' => '1473428717_6-577591_garage_677-302-782-534_2396_869.jpg',
        'instanceName' => 'garage',
    ]

To solve this there are two ways:
* create your own proxy webhook: 
    * define the address of your own webhook in the Io Device.
    * when you webhook receives the POST request from Kerberos.io, you can create your own request to the Slack API.

* Or you can modify the machinery yourself, so that it contains a text property.

Hope This helps @ypwu1

ypwu1, 2016-09-09:
Thanks for that, I am writing a proxy to handle it.

closed 2016-09-09 by ypwu1

Moulde, 2018-02-25:
Wouldn't it be easier to write a python script that sends the request to the slack webhook instead of writing a proxy?

cedricve, 2018-02-25:
@Moulde, well that would also be some kind of proxy no?

Moulde, 2018-02-25:
Yes that's correct. 
But then you would not use the webhook setting in kerberos, but instead the script function.

cedricve, 2018-02-25:
correct @Moulde

Moulde, 2018-03-25:
So python runs, but I need a few packages (slackweb and dropbox), but pip is not installed, and as mentioned, file system is read-only, so it's not really easy to install either.

Do I need to do raspbian install for this to work?

cedricve, 2018-03-25:
Yep indeed, KiOS is more or less a production version. If you want to do development, you would rather use another OS like Raspbian.