kerberos-io/machinery #56: IO: can't execute error

HanYangZhao, 2016-09-20:
Hello,

I'm trying to run kerberos with 2 IO devices, one is a disk on a Pi and the second one is a Webhook. However i'm only able to detect one event before start running into IO: can't execute error. Here is the log msg

18:51:28.884 INFO  [trivial] Starting io devices: Disk,Webhook 
20/09/2016 14:51:45.810 INFO  [business] HullExpositor: activity detected from (327,491) to (406,576)
20/09/2016 14:51:45.815 INFO  [business] Heuristic is valid; numberOfChanges: 1133, Sequence duration: 1
20/09/2016 14:51:46.271 INFO  [business] Executing IO devices for 1 detection(s)
20/09/2016 14:51:46.276 INFO  [business] IoDisk: saving image tower_1474397505_327-491-406-576.jpg
20/09/2016 14:51:46.408 INFO  [business] IoWebhook: post to webhook 10.42.0.1:3000/cameraevent
20/09/2016 14:52:00.022 INFO  [business] HullExpositor: activity detected from (325,484) to (406,575)
20/09/2016 14:52:00.027 INFO  [business] Heuristic is valid; numberOfChanges: 889, Sequence duration: 1
20/09/2016 14:53:46.424 ERROR [trivial] IO: can't execute
20/09/2016 14:53:49.932 INFO  [business] Executing IO devices for 2 detection(s)
20/09/2016 14:53:49.937 INFO  [business] IoDisk: saving image tower_1474397505_327-491-406-576.jpg
20/09/2016 14:53:50.070 INFO  [business] IoWebhook: post to webhook 10.42.0.1:3000/cameraevent
20/09/2016 14:55:50.086 ERROR [trivial] IO: can't execute
20/09/2016 14:55:50.594 INFO  [business] IoDisk: saving image tower_1474397505_327-491-406-576.jpg
20/09/2016 14:55:50.722 INFO  [business] IoWebhook: post to webhook 10.42.0.1:3000/cameraevent

I'm using the DifferentialCollins with a erode of 5 and threshold of 25 , Sequence Heuristics with a minChange : 20 , minDuration : 1 , noMotionDelayTime : 2500

Any help would be appreciated

cedricve, 2016-09-20:
@HanYangZhao did you tried to run them separate? So we can see at which IO it goes wrong..

cedricve, 2016-09-20:
Can you confirm that the webhook returns a HTTP 200?
https://github.com/kerberos-io/machinery/blob/master/src/kerberos/machinery/io/IoWebhook.cpp#L45

HanYangZhao, 2016-09-20:
thanks turns out that webhook wasn't returning a proper HTTP 200.

closed 2016-09-20 by HanYangZhao