kerberos-io/machinery #57: Face detection and smart alarms

marktheunissen, 2016-10-08:
Hi again, are there any plans to do smarter video analytics, the kind seen on the latest cameras? For example: line crossing detection, face detection, left object detection, etc?

cedricve, 2016-10-08:
Hi

Yes we already have a counter heuristic, which counts incoming and outgoing objects (=line crossing). Face recognition will be integrated later.

https://doc.kerberos.io/2.0/machinery/heuristics/counter

For the moment we are focussing on alerts, e.g. the ability to categorise specific activity (my kids left the house with their bike, my wife is washing the car, etc.). The idea is that we are getting closer to the area of machine learning (a self learning video surveillance systems); an announcement will follow. However these functionalities will only be available on our cloud application (as it requires quite some CPU and memory). 

Regards
Cédric

marktheunissen, 2016-10-08:
That sounds great. Are you planning on something like https://canary.is?  I also noticed that Tensorflow runs on Raspberry Pi, which promises some cool object detection and classification.

cedricve, 2016-10-29:
@marktheunissen smart alarms (notifications) will be introduced in a next release of the cloud application. We will add machine learning, and the possibility to tag events (which are used in the AI process).

closed 2016-10-29 by cedricve

SudoPlz, 2020-11-02:
Is that any closer to being a thing now (2020) ?