kerberos-io/machinery #58: UI Suggestions + Counter&Hull Bug.

nachiarg, 2016-10-12:
Hi, Ive been using Kerberos for a couple of days and I`m really pleased with the software . It is really stable with a great UI. 
Ive made a suggestion list based on my experience as a user:

1- Can you increase the capture times, my RPI2 + RPIcamera is capturing 1 frame every 5 seconds (I reduce every Delay time to 0)
2- Introduce a calendar view. Each day will have the number of events, pictures, incoming and outgoing stats.
3- Introduce a list view for slower connections, not just thumbnails.
4- Include the Incoming and outgoing stats to the dashboard.
5- Zapier / IFTTT channel? (currently using webhook workaround)
6-Different triggers/webhooks for Incoming/outgoing and regular events
7-Can you code to use the RPIcamera led as an indicator of an event being registered?

Also I found a Bug with the Hull & counter UI. The camera capture for drawing lines or hull are displayed using a responsive image so you end up with lines pipes locked in the wrong coordinates. 

Keep it up!

Juan

cedricve, 2016-10-12:
Hello Juan

Thank you for awesome feedback, this is highly appreciated! Please find my feedback below.

1- Can you increase the capture times, my RPI2 + RPIcamera is capturing 1 frame every 5 seconds (I reduce every Delay time to 0)

My first thought is that you're processing HD images, can you confirm the resolution you've defined?
It looks like the RPi can't follow. Additionaly can you send me your system configuration (which you can download from the settings page) and send it to hello@cedric.ws

2- Introduce a calendar view. Each day will have the number of events, pictures, incoming and outgoing stats.

Nice idea, I think I can imagine what you mean, but I'm not really convinced about the benefit.  It would be helpful to have a summary of intelligent notifications. 

E.g. 14th of April
9:00 Juan went to work
9:00 Children go to school
10:30 Mailman came by
etc..

The nice thing is that we are working on this and probably can integrate this in the cloud application. More news to come..
https://kerberosio.zendesk.com/hc/en-us/articles/212180749-Browser-Notifications

3- Introduce a list view for slower connections, not just thumbnails.

Great idea, what would you show as data. Just the timestamps with the number of images?

4- Include the Incoming and outgoing stats to the dashboard.

You mean when using the counter heuristic?

5- Zapier / IFTTT channel? (currently using webhook workaround)

Well we plan not to do this on the RPi, so your webhook workaround is great. We will add this to our smart notification system on the cloud application (released in some weeks/months)

6-Different triggers/webhooks for Incoming/outgoing and regular events.

Ok, sounds useful.

7-Can you code to use the RPIcamera led as an indicator of an event being registered?

I think so, but didn't tried. Could be a nice feature :+1:  great idea.

Also I found a Bug with the Hull & counter UI. The camera capture for drawing lines or hull are displayed using a responsive image so you end up with lines pipes locked in the wrong coordinates.

Not 100% clear what you mean, can you explain more in detail?

Cédric

nachiarg, 2016-10-12:
wow, what a quick feedback

1- Im using 800x600 . Uploaded config files.
[kerberosio-configuration.tar.gz](https://github.com/kerberos-io/machinery/files/524787/kerberosio-configuration.tar.gz)

2- I uploaded an example from google images . Its a calendar view for events. Easier navigation and quick look
[image]

3- Exactly, a simple scrolldown list to avoid image loading.

4- Yes. Im using your webhook to log it outside kerberos but it could be easily tracked inside your app.

5,6,7- Ok.

8- When you resize the web browser coordinates for the lines change but image is displayed in full width so you end up with wrong coordinates being saved. If you resize an image the lines follow the action but the coordinates vector change; this means counter lines or hull polygon is saved in the wrong region. 

Juan

cedricve, 2016-10-13:
Hello Juan

1. Well it looks like the RPi can't follow, you should try a Raspberry Pi 3, although I don't think that will matter a lot. The counter heuristic + backgroundsubtraction is a very computational operation. You could experiment with the differential images algorithm instead. Let me know how it goes.

8. Which browser are you using, I programmed it against Chrome and it works properly. The coordinates are scaled against the size of the images.

Cédric

cedricve, 2016-10-17:
We've added the timeline/calendar feature to the cloud application. We'll see if we can move it to the web interface as well.

[image]

nachiarg, 2016-10-17:
Looks great. I couldn't find  information about the cloud service (pricing, plans , features) . Could you point me in the right direction?

Thanks

cedricve, 2016-10-17:
If you signup at cloud.kerberos.io and signin you'll see the pricing model. Hope this helps

closed 2016-12-11 by cedricve