kerberos-io/machinery #59: File capture

JonasGroeger, 2016-12-01:
Hello!

I'm writing you because I have a use case in which the videos were taken without Kerberos. Now, I would like to see the movement in the videos without capturing from a camera but by just using the file.

Of course it has to do this faster than the capture, since the whole data is already in the file.

Looking at the source, I see that there is no FileCapture setting nor code for it. So I'm assuming this is not possible right now.

What would be the steps to have this and how would a contribution look like?

cedricve, 2016-12-01:
Hello @JonasGroeger 

Yes it is possible, we have the VideoCapture device. You can specify the path to the video file.

[image]

Hope this helps

JonasGroeger, 2016-12-01:
@cedricve Is the video then played at normal speed or sped up?

cedricve, 2016-12-01:
It depends on the machine you're running Kerberos.io and the resolution of the video. If kerberos.io can process faster than the FPS of your video, than yes otherwise no.

What are the specs of your video file? (FPS and resolution)

JonasGroeger, 2016-12-01:
@cedricve Its a 1280x720 with 30FPS (15FPS, frames are doubled). I would be running Kerberos on a quite powerful machine.

cedricve, 2016-12-01:
To be honest don't think you will be able to go faster than the video, with Kerberos.io.

cedricve, 2016-12-01:
The difference between video files and a capture device (usb camera), is that each frame in the videofile will be processed. With a usb camera the frames which are stored in the buffer of the camera are processed. The usb camera clears it buffers so it shows nearly real time images.

cedricve, 2016-12-10:
hey @JonasGroeger any update on this?

JonasGroeger, 2016-12-11:
@cedricve The use case I had is no longer a thing, unfortunately.

closed 2016-12-11 by JonasGroeger