kerberos-io/machinery #60: Enhancement request - Output as video file

vgoncearencu, 2016-12-02:
Hello,
it would be great to have possibility if a motion detection is triggered to save that event not as a series of images but instead trigger a video record and save it to a file with possibility to view it from WebUI .

So from configuration page we could choose, output type as either:
- Image (snapshot)
or
- Video record

Thank you,
Victor

marksev1, 2016-12-06:
Maybe the images could still be used but in the WebUI there would be a slider to fast scroll through the images? Thus creating a video with sliding the slider faster :).  What do you think?

cedricve, 2016-12-06:
Hello @vgoncearencu 

I've thought about this feature a lot, and the reason why I didn't developed is because I wasn't able to save the videofile with the correct FPS (performance of the machine to process the algorithms). It's a hard exercise, and requires a different approach..

But indeed it would be a nice feature.
Cédric

cedricve, 2016-12-06:
@marksev1, problem is that loading all the images separately will request more bandwidth. At this moment the first 3 images are preloaded. I admit if there are a lot of images, it's a repetitive task.

cedricve, 2016-12-07:
@vgoncearencu @marksev1 I'll see if I can make develop something clean for this. Probably I'll need help from an expert.

cedricve, 2016-12-11:
Currently implementing the video feature, will release a develop version once I have a stable solution.

vgoncearencu, 2016-12-11:
thank you guys you are the best!

cedricve, 2016-12-16:
Video's will be displayed in the GUI like this, what do you think? A player will open once you've clicked on a camera icon.

[image]

vgoncearencu, 2016-12-18:
Looks good. The player will have a start/stop/pause buttons ?

cedricve, 2016-12-18:
yes @vgoncearencu 

[image]

However when multiple videos are recorded as visualised above I would like to have a playlist as well. what do you think? 

[image]

vgoncearencu, 2016-12-18:
yes. That will make watching videos much easier.

cedricve, 2016-12-21:
Update on player, now includes a playlist.

[image]

marksev1, 2016-12-21:
Its soo cool to watch the developement in realtime Cedric! :D

cedricve, 2016-12-29:
Finished this feature, and added to develop branch. This will be integrated in next release. 

Features:
- Video and or Image can be saved.
- When only video's are saved, the first video in a sequence will be play automatically in the web interface.
- When more videos are saved in a sequence a playlist is available.
- Responsive support.

Results:

[image]

[image]

cedricve, 2017-01-08:
will be integrated in version 2.2.0

closed 2017-01-08 by cedricve