kerberos-io/machinery #61: Does it run on other SBCs?

marksev1, 2016-12-06:


cedricve, 2016-12-06:
Hey :) Yep you can follow the Raspbian installation for the Orange Pi.

marksev1, 2016-12-06:
But which Rpi version would i choose so that I get the most correct .deb, should i use RPi2 or RPi3 deb for example?

cedricve, 2016-12-06:
What are the specs of the OrangePi?

marksev1, 2016-12-06:
Its Allwinner H3 cpu -> https://linux-sunxi.org/H3

Specs of Orange Pi PC board:
Specifications
SoC	H3 @ 1.3 GHz
DRAM	1GiB DDR3L (v1.2 K4B4G1646Q-HYK0, v1.3 K4B4G1646E-BYK0)
Power	DC 5V @ 2A (4.0mm/1.7mm barrel plug - centre positive)
or via GPIO header pins
Features
Video	HDMI (HDCP, CEC), CVBS
Audio	3.5 mm Jack, HDMI, Microphone
Network	10/100 Ethernet
Storage	µSD, 8GB eMMC on PC Plus
USB	3 USB 2.0 Host, 1 USB 2.0 OTG
Other	IR

cedricve, 2016-12-06:
You can use the Raspberry Pi 2 debian file, give it a try.

marksev1, 2016-12-06:
Need to first buy an IP-cam, but when I do I will for sure try it cedricve! Thanks! Will let you know how it goes and how we can later on add support for it if it doesn't work on first try.

closed 2016-12-11 by cedricve