kerberos-io/machinery #62: Request - add "alarm scheduling and mail delivery of "event" picture

marksev1, 2016-12-06:
A little bad title to the issue but here it goes.
I

cedricve, 2016-12-08:
Thanks for the feedback, this is indeed an interesting feature. Let's collaborate on how you think this would fit best with other technologies. By the way can add new features in the feature list http://feathub.com/kerberos-io/machinery.

Thank you for your motivation and enthusiasm!

marksev1, 2016-12-08:
Well the bigger players in the open-source home automation game are Node-red, Home Assistant, Domoticz and Openhab...And they all support MQTT. Its really a nice protocol, so basically if you just supported pushing the "motion-detection-events" via MQTT, it could be same json.

And in the dashboard just one more text input box to input the MQTT topic to which you would publish the json.

Yup I'm very excited about your project!

cedricve, 2016-12-08:
Great idea I like it. Can you think of possible use cases and benefits in how you see this working? E.g. Openhab and Kerberos.io

cedricve, 2016-12-08:
@marksev1 added feature to http://feathub.com/kerberos-io/machinery

marksev1, 2016-12-08:
Well with Node-red for example you could for example set up (Tweet me if at night someone is detected, or send me an email or send me an sms or pushbullet/pushover/etc notification or even sound an alarm)? You could for example use "sensor fusion" to have a more firm grip if someone is really inside (camera motion detected + PIR sensors). 
Or you could trigger the timelapse mode and later on disengage it at specific time-schedules in a day for some nice videos.

Or if you added support for PTZ in cameras inside kerberos.io you could control it remotely via the node-red-dashboard. Or even if you wouldn't add support for PTZ you could still control it directly from node-red. 

Basically the sky is the limit :).

With Openhab you could do that also but I don't have enough experience with it to whip it up for it specifically.

cedricve, 2016-12-08:
Hmm interesting, I assume you have a Node-red setup available? Can you provide me with some clear information/tutorial, I think you know best what are the requirements. And most important once developed can you test it?

marksev1, 2016-12-08:
This nodered comes bundled with Raspbian thats why its an interesting option for your users ->  http://nodered.org/ here are the getting started guide and stuff..In addition to this you also need the mosquitto mqtt broker running on the RPi. Then onward its very simple, drag and drop. 

Yes I can test it the mqtt thing, but first need to buy some stuff, so if you can simulate a motion detection even that I would just test the protocol part it would be good:).

But I think now during the holidays I will buy myself some stuff :).

muten84, 2016-12-08:
If interested i integrated kerberos.io with an MQTT broker using a node.js script that listen to a TCP socket that react to the TCP IO trigger provided  by kerberos.io. The MQTT broker is mosquitto and  it serves as broker for all my IOT devices such as pir sensors, and proximity. The base board i used for IOT is the WeMos D1mini that implements Arduino API. MQTT was designed to be fast and scalable when using sensors that produces a lot of messages. Keep in mind that using a broker has some cons since it is the single point of failure of your middleware.

marksev1, 2016-12-08:
Interesting, but would probably still be better if a native implementation existed?

cedricve, 2016-12-08:
@muten84 nice I wasn't aware that you advance so far, great job. However I don't think less-advanced users, will not be able to create something like you've did. I would suggest to follow @marksev1 approach so it's available for a broader segment?..

muten84, 2016-12-08:
@cedricve i totally agree, non skillled users should have a built in feature what i meant is that i started with the TCP io trigger to have a POC of the MQTT feature and to realize what kerberos should really  provides. With the actual scenario an MQTT client configuration should be enough. The client has to know the ipband the port of the broker and the dest topic for publishing messages. On the other side there is the opportunity to provides a functionality to subscribe kerberos to an MQTT broker for triggering some functions like service restart and Image removal, basically a set of useful capabilities, but i think that this second scenario is more far from kerberos.io purposes.

cedricve, 2016-12-08:
Great news, do you have this open sourced? Well the second scenario, can be done through REST API calls to the kerberos.io instance. This isn't documented but we've added this a long time ago. More information here: https://github.com/kerberos-io/web/blob/master/app/api.php#L72-L93.
Calls are secured with basic authentication (username and password to sign-in).

muten84, 2016-12-08:
I can copy and paste the script here, the instance variable should be injected from kerberos when trigger events so the script become stateless. A bit of doc the cript create a tcp server once created connect to mqtt broker. Every time the tcp server receives a packet it forward on apublish on the mqtt client. This is a code snippet and should be tested, my original script is more complex and and it cannot be shared in that condition :)

```javascript
var net = require('net');
var MQTTClient = require('MQTTClient').Client;

var instance = "Camera1";
var connected = false;
var serverPort = 54321;

var options = {
	client_id:	'kerberos.io_'+instance;
}

var client = new MQTTClient('localhost', 1883, options);
	
function publish(data){
    if(connected){
      client.publish('home.'+instance, data, options, function (message_id) {
                console.log("message published");
      });
    }
}

net.createServer(function (socket)
{
    client.connect(function () {
      	console.log("connected");
        connected = true;
     });
     
    // Handle incoming messages from the magnet controller.
    socket.on('data', function (data){
          publish(data);
    });
}).listen(serverPort);
```

REST API is for me a great info!! Thanks!

muten84, 2016-12-08:
I just realized that with the IFTT app we can create an applet that receives a web request from a Maker service (provided by IFTT) and triggers an Android Wear notification (if you have a Smart Watch) ..... the web request could be triggered from node-red that is subscribed to the mqtt broker. The mqtt publish is triggered from the script above :). No code only tools, the choice is trigger a web request directly from kerberos.io,  using an mqtt,  or both? :)

muten84, 2016-12-08:
@cedricve  is there a way from the REST API to obtain the last saved image URL?

marksev1, 2016-12-09:
Muten84 so you mean IFTT just as an external service, after the mqtt transfer to node-red was already made? I can make node-red also text me or send me a google hangouts notification or even email me with the picture..

Got another off-topic idea, would it be possible, if one has multiple cameras and multiple instances with some tool to create a big picture out of the last pictures of all the cameras? :) I'm sure you muten84 could do it :P

muten84, 2016-12-09:
@marksev1 Yes node-red is a powerful tool the purpose of my POC is to unserstand what feature kerberos.io really needs to provide. I shared the POC state and i think that MQTT client can just be used from node Red and kerberos web hooks can be used to provides a set of nodeRed nodes based on kerberos capabilities. NodeRed has a contrib project page where everyone can deploy its set of nodes... maybe i can deploy mines :D. If so MQTT client becomes needless.

muten84, 2016-12-09:
`
{ "regionCoordinates": [ 507, 271, 607, 350 ], "numberOfChanges": 1141, "timestamp": "1481283474", "microseconds": "6-934128", "token": 232, "pathToImage": "1481283474_6-934128_Zona1_507-271-607-350_1141_232.jpg", "instanceName": "Zona1" }`

UPDATE: this is the json sent with a webhook POST Request, the json become a message in nodered and can be processed by a node red function node or by a custom nodered node. There are some useful built-in function nodes such as split to split messages, join function to buffer a sequence of messages.....you can create custom prototye and POC with all provided nodes.

I will be glad to share some useful flows on nodered page. I suggest this approach because i think that a built in function should be generic (like webhooks, mqtt clients, tcp triggers) or specific but useful to most users. We can see node red flows as the incubator of new functionality for kerberos.io.

@cedricve  What do you think  about ???

cedricve, 2016-12-09:
@muten84 That would be great, let's start by defining some usecases with how we could integrate Kerberos.io with node red, and how it will work in practice (my knowledge is too limited in this domain so I'll need feedback and clear instructions from you guys @muten84 @marksev1.

My idea at this moment is to create a new IoDevice -> IoMQTT, which converts the json object to the IoMQQT protocol.

muten84, 2016-12-09:
Work in progress: when i will finish i will document better this simple flow. A little preview: i'm building a simple flow to split the kerberos.io output when there are too much notifications. The flow reacts  when a webhook is trigerred so every time kerberos collect a new image the flow wait for 5 seconds, if no events comes after this timeout it will proceed to request an http POST to the IFTT maker service then the applet installed on my own IFTT profile react to send the notification to my smart watch sending me a preview. The IFTT maker support a json object with prefixed name such as Value1, Value2, and so on. The processData function fill the Value1 with image path so the IFTT service can send me the image preview on my smart watch.
[image]

cedricve, 2016-12-10:
Ok cool, can this follow be packaged and provided to the node red community, if yes this would be game changer! @marksev1 I don't know if other home automation systems could provide the same functionality, e.g. OpenHab?

Would it still be useful to implement the MQTT protocol? We should list the most popular home automation service and the input protocols they support (Webhook, MQTT, etc.)

muten84, 2016-12-10:
@cedricve yes of course, at [this ](http://flows.nodered.org/) link there are many flows and nodes that everyone can download and reuse, everyone can upload yur own, the same in IFTT for smartphone integration.

cedricve, 2016-12-10:
@muten84 yes you can retrieve the last image from the API by calling the latest sequence end point. https://github.com/kerberos-io/web/blob/master/app/api.php#L57

However I think that you will need to duplicate it, and move it under the auth.basic group. https://github.com/kerberos-io/web/blob/master/app/api.php#L72-L74

cedricve, 2016-12-10:
@muten84 great, than I propose to use your flow for node red in production, and document it on https://doc.kerberos.io. What do you think? This will be then the way to go, if you want to connect kerberos.io with nodered.

Regarding other home automation tools, we should consider to implement the MQTT.

muten84, 2016-12-10:
I totally agree. Is _"Addons"_ section the right candidate for this kind of documentation? Did you find an MQTT lib in C++ for publishing messages?

cedricve, 2016-12-10:
Yes indeed, we can work on it together once you have a stable and published version of your workflow. I'll look for a MQTT library, and come back to you with a possible solution. Thanks @muten84 !

marksev1, 2016-12-10:
I found this libs not sure how good they are :)

https://eclipse.org/paho/clients/c/
https://eclipse.org/paho/clients/cpp/

marksev1, 2017-05-08:
So any new progress on this front? (its my number one wish hehe :D)

cedricve, 2017-08-21:
Follow-up ticket here https://github.com/kerberos-io/machinery/issues/89

closed 2017-08-21 by cedricve