kerberos-io/machinery #64: Weird colors from RPi Camera V2, possible white balance adjustment?

HanYangZhao, 2016-12-19:
The colors from the stream and images are weird. For some reason, the skin color is blue. This happens both with the NoIR V2 as well as the normal V2. Perhaps it's a white balance issue?

Thanks

cedricve, 2016-12-20:
Are you using the raspbian images?

HanYangZhao, 2016-12-20:
Yes I also tried compiling the latest version of the master branch but the results were the same

cedricve, 2016-12-20:
Did you upgrade to latest firmware?

HanYangZhao, 2016-12-20:
Yes did a sudo apt update and upgrade before compiling

HanYangZhao, 2016-12-20:
3

Whytehorse, 2016-12-23:
The V2 camera is much more sensitive than V1. You can tune the parameters in the uv4l conf file. I forget the name if it

cedricve, 2016-12-26:
@HanYangZhao a bug in the raspicam library we are using, we will integrate the fix in next release.

HanYangZhao, 2016-12-29:
Thanks. Is there a quick fix I can apply right now?

closed 2017-02-13 by cedricve