kerberos-io/machinery #66: undefined reference to `mmal_util_rgb_order_fixed'

mishab, 2017-01-10:
Tried building the project and got this error message:

/home/pi/machinery/build/thirdparty/lib/libraspicam_cv.a(private_impl.cpp.o): In                                            function `raspicam::_private::Private_Impl::create_camera_component(raspicam::_                                           private::RASPIVID_STATE*)':
private_impl.cpp:(.text._ZN8raspicam8_private12Private_Impl23create_camera_compo                                           nentEPNS0_14RASPIVID_STATEE+0x58): undefined reference to `mmal_util_rgb_order_f                                           ixed'
collect2: ld returned 1 exit status
src/CMakeFiles/kerberosio.dir/build.make:562: recipe for target '../bin/kerberos                                           io' failed
make[2]: *** [../bin/kerberosio] Error 1
CMakeFiles/Makefile2:1140: recipe for target 'src/CMakeFiles/kerberosio.dir/all'                                            failed
make[1]: *** [src/CMakeFiles/kerberosio.dir/all] Error 2
Makefile:143: recipe for target 'all' failed
make: *** [all] Error 2

cedricve, 2017-01-10:
Did you updated / upgraded your Raspbian version?

mishab, 2017-01-10:
I did "sudo apt-get update && sudo apt-get upgrade". Anything else I might be missing?

cedricve, 2017-01-10:
Can you provide me with more logging?

mishab, 2017-01-10:
Would be happy to provide logging. What do you need? Meanwhile I nuked the directory and restarted building everything from scratch

cedricve, 2017-01-10:
Looks like you don't have the latest mmal libraries installed..

https://github.com/raspberrypi/userland/blob/master/interface/mmal/util/mmal_util.h

cedricve, 2017-01-10:
Van you verify the mmal_util_rgb_order_fixed method is in place?

mishab, 2017-01-10:
I thought cmake will do it for me(External-RaspiCam.cmake) and pull the raspicam repo ( ${git_protocol}://github.com/cedricve/raspicam )

cedricve, 2017-01-11:
Weird, can you try to build this repository directly https://github.com/cedricve/raspicam?
On which RPi are you trying to build?

cedricve, 2017-01-12:
@mishab can you show me the version of your cmake?

pi@raspberrypi:~/machinery/build $ cmake -version
cmake version 3.6.2

mishab, 2017-01-15:
same. 3.6.2
do you know what causes this build error? what version of ffmpeg is required?

[ 77%] Linking CXX executable ../../bin/kerberosio
../thirdparty/lib/libopencv_videoio.a(cap_ffmpeg.cpp.o): In function `CvVideoWriter_FFMPEG::open(char const*, int, double, int, int, bool)':
cap_ffmpeg.cpp:(.text._ZN20CvVideoWriter_FFMPEG4openEPKcidiib+0x2dc): undefined reference to `avformat_get_mov_video_tags'
collect2: error: ld returned 1 exit status
src/CMakeFiles/kerberosio.dir/build.make:594: recipe for target '../bin/kerberosio' failed
make[2]: *** [../bin/kerberosio] Error 1
CMakeFiles/Makefile2:1187: recipe for target 'src/CMakeFiles/kerberosio.dir/all' failed
make[1]: *** [src/CMakeFiles/kerberosio.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
pi@raspberrypi-3:~/machinery/build $ cmake -version
cmake version 3.6.2

cedricve, 2017-01-16:
I confirm that there is an issue on ARM if you have FFMPEG installed. Looks like the `avformat_get_mov_video_tags`function is missing. However when you just install libav-tools on your RPi it should work (without compiling ffmpeg). You should try to remove the FFMPEG installation with all extensions, and just rely on libav-tools.

mishab, 2017-01-17:
Cedric, 
I was able to figure everything out, except for VideoCapture. I am using Raspberry Pi 3, but when I configure kerberos to use VideoCapture, OpenCV errors out with the following error:
GStreamer Error opening bin 0.
I see that the VideoCapture path is 0, which seems to be the correct index for my USB camera.
Any idea why OpenCV can’t open camera at index 0, while ffmpeg is capturing video just fine:
pi@raspberrypi-3:~/machinery/build $ sudo ffmpeg -f v4l2 -framerate 25 -video_size 1920x1080 -i /dev/video0 /etc/opt/kerberosio/capture/output.mp4

cedricve, 2017-01-17:
You need to select USBcamera, instead of Videocapture. The videocapture is for debugging purposes, you can specify a video file, and let it processing by Kerberos.io just like it was a real-time capture device.

closed 2017-01-31 by cedricve

touchardv, 2017-03-19:
Hi,
I recently tried to also upgrade my Pi with the latest version of Kerberos (great job btw) and also had an issue with libmmal (missing symbol: `undefined symbol: mmal_util_rgb_order_fixed`).
I ran all updates/upgrade, but the lib is still the same... How is it supposed to be updated on Raspbian?

cedricve, 2017-03-19:
@touchardv which version did you installed, on which Pi?

touchardv, 2017-03-20:
That was machinery version 2.2.0 and the Pi is the 2 model B running on:
4.4.0-1-rpi2 #1 SMP Debian 4.4.6-1+rpi14 (2016-05-05) armv7l GNU/Linux

Tried update/upgrade/dist-upgrade but nothing seemed to change that `mmal` library.

cedricve, 2017-03-20:
Did you already tried on a fresh installation?

touchardv, 2017-03-20:
I did not try a fresh installation. The Pi was re-installed beginning of January 2017 with a fresh Raspbian network installation.

cedricve, 2017-03-21:
@touchardv just tried out myself with a Pi2 model B, and cannot reproduce with latest raspbian installation..

touchardv, 2017-03-21:
I think I know what is wrong: the image I used: https://github.com/debian-pi/raspbian-ua-netinst
I used the release 1.0.9 there and it looks a bit old. But then, still wondering, how can one update this libmmal?

cedricve, 2017-03-28:
@touchardv  any update?

touchardv, 2017-03-29:
I am planning to re-re-install the Pi when I have time, probably in the coming days. I ll post an update then.

touchardv, 2017-04-19:
I can confirm that the latest release (2.2.1) works just fine on a fresh Raspbian installation (Raspbian Jessie Lite from https://www.raspberrypi.org/downloads/raspbian/).
So clearly there is something in the previous version I was using (https://github.com/debian-pi/raspbian-ua-netinst) that was installing an old version of this library... I guess?

Anyway, thanks for checking and support.