kerberos-io/machinery #67: RTSP ip camera - capture issue

mcgyver83, 2017-01-14:
Hi all, I'm running kerberos.io on a raspberry pi2 but I cannog get my Sricam ONVIF ipcam working with RTSP streaming.
This is working with vlc and rtsp://192.168.0.101:554/onvif1 but I cannot get it working with kerberos :(

```
04/01/2017 22:03:48.714 INFO  [trivial] Starting capture device: McG-camera
04/01/2017 22:03:48.715 ERROR [trivial] Kerberos : Factory : Could not create an instance of message: McG-camera
```

```
<streams>
    <Rtsp>
        <enabled type="bool">true</enabled>
        <streamPort type="number">554</streamPort>
        <quality type="number">75</quality>
    </Rtsp>
</streams>

<captures>
    <IPCamera>
        <url type="text">rtsp://192.168.0.101:554/onvif1</url>
        <frameWidth type="number">640</frameWidth>
        <frameHeight type="number">360</frameHeight>
        <delay type="number">1000</delay>
        <angle type="number">0</angle>
    </IPCamera>
</captures>
```

cedricve, 2017-01-14:
Hey @mcgyver83 

Looks like you've changed the config files a little bit too much. You should not touch the XML tags, only the contents.

 [trivial] Starting capture device: McG-camera

You should keep the name IPCamera in the config.xml file (it's linked internally) + you should not touch the MJPG tag (this is for the MJPEG broadcast of kerberos.io)

mcgyver83, 2017-01-14:
Many thanks...I made too many changes 👍 
Now no errors in log.
Now I have another problem but I'm going to open another issue.
Can I ask support using another site instead of issues in github?

closed 2017-01-16 by cedricve

djmgrant, 2020-11-15:
Im getting a similar error :   

14/11/2020 19:11:38.915 INFO  [trivial] Logging is set to info
14/11/2020 19:11:38.916 INFO  [trivial] Capture: Start capture device: RaspiCamera
14/11/2020 19:11:38.916 ERROR [trivial] Kerberos : Factory : Could not create an instance of message: 
14/11/2020 19:11:41.916 INFO  [trivial] Reading configuration file:

Im using a raspi camera :

Config
<?xml version="1.0"?>
<kerberos>

    <instance>
        <name type="text">hall</name>
        <logging type="bool">false</logging>
        <timezone type="timezone">Europe-London</timezone>
        <capture file="capture.xml">RaspiCamera</capture>
        <stream file="stream.xml">Mjpg</stream>
        <condition file="condition.xml" type="multiple">Enabled</condition>
        <algorithm file="algorithm.xml">DifferentialCollins</algorithm>
        <expositor file="expositor.xml">Hull</expositor>
        <heuristic file="heuristic.xml">Sequence</heuristic>
        <io file="io.xml" type="multiple">Video</io>
        <cloud file="cloud.xml">S3</cloud>
    </instance>

</kerberos>