kerberos-io/machinery #68: Docker image for raspberry pi

dragoncube, 2017-01-24:
It would be very nice to have docker image for raspberry pi (using with Hypriot OS, for example). It would allow to side-load some extra stuff (like exposing temperature sensors attached to pi via REST and so on)

cedricve, 2017-01-25:
Hi @dragoncube 

Well I don't think that will be hard to develop, however the thing I'm concerned about is automated builds. At the moment we have Travis-CI building the docker images for x86, each time a new commit has been pushed. I'm not aware of something similar for ARM machines?

dragoncube, 2017-01-26:
@cedricve yes, automated builds a problem, you either need to have ARM build agents, or deal with QEMU, which is that straightforward (https://resin.io/blog/building-arm-containers-on-any-x86-machine-even-dockerhub/). Though it would be good at least to have verified Dockerfile which you can build locally on your own pi.

cedricve, 2017-02-10:
I will work on this @dragoncube, if you can help that would be great ;)

dragoncube, 2017-02-14:
@cedricve I definitely can help with testing. BTW, I've went through the article once again, looks like it should be straightforward:

`
FROM resin/armv7hf-debian

RUN [ "cross-build-start" ]

RUN apt-get update  
RUN apt-get install python  
RUN pip install virtualenv

RUN [ "cross-build-end" ] 
`

cedricve, 2017-02-15:
This might be more interesting: http://blog.hypriot.com/post/setup-simple-ci-pipeline-for-arm-images/

cedricve, 2017-03-05:
Discussion forwarded to docker repo.  Multi-architecture support:  kerberos-io/docker#3.

closed 2017-03-05 by cedricve