kerberos-io/machinery #69: Request - GPIO Input to enable/disable motion detection

HanYangZhao, 2017-02-10:
I know there is currently GPIO output when motion is detected , but I was wondering if a GPIO input to enable/disable the "enable" parameter could be implemented. One use case that I'm thinking of is to use a PIR motion sensor to enable motion detection only when humans or animals are passing by and ignore things such as cars,trees.... Would appreciate if someone can give me some hints on where to start.

Thanks

cedricve, 2017-02-10:
Hey @HanYangZhao, I've been thinking about it before however I didn't implement it at that time because it's a big change request, and I had the feeling that it would become more than a video surveillance system.

If we would decide to implement this, we will also need to support other channels, e.g. incoming webhooks. At the moment of writing I see following input and output devices.

## Inputs

* Capture device
* GPIO pin (new)
* Webhook (new)

## Outputs

* Video
* Snapshot
* Webhook
* GPIO
* TCP client

HanYangZhao, 2017-02-16:
That sounds pretty good. I think someone was already working on the webhook input #65 ?

cedricve, 2017-03-13:
@HanYangZhao yep that's correct. You can simulate it by using the REST API and your custom script. https://doc.kerberos.io/2.0/web/aPI