kerberos-io/machinery #71: IP camera stops grabbing images

cedricve, 2017-02-16:
With the new version of Kerberos.io 2.2.0 (OpenCV3.2.0), the ip capture device stops grabbing images at some time. Needs to be investigated.

muten84, 2017-02-16:
I have two instances of 2.2.0 both configured ti capture from IP camera and i have the same issue ... The strange thing is that happens only on One instance ... The same instance between 00:00 and 2:00 AM. Hope this helps.. i have to investigate more deep

cedricve, 2017-02-17:
This issue have been resolved, I'll create a new version asap.
https://github.com/kerberos-io/machinery/commit/a040864ba4ca0b1f1780de7e13eb14f3278fc97d

cedricve, 2017-02-18:
Please reinstall 2.2.0 I've updated the image, if this is stable again, I'll create a 2.2.1 release. Keep me posted @muten84

closed 2017-02-18 by cedricve

muten84, 2017-02-19:
i reinstalled release from the deb package .... waiting for a result within a couple of days. I have two instances ... only one of them use new version. I hope the deb i installed contains the fix, i'm not sure about that.

cedricve, 2017-02-19:
Not sure if I replaced the .deb files, however I did it now. So if you will pull the .deb again, it will contain the fix.

muten84, 2017-03-02:
Ok seems to be working now