kerberos-io/machinery #74: Image lock

nragon, 2017-02-21:
Hi,

Image is getting locked until `cv::Mat m_image` is out of scope. It's impossible to use the capture(open in rb mode) after IoDisk. Maybe you can deep copy m_image and save its copy, just guessing.

Thanks

cedricve, 2017-02-21:
@nragon can you reference the code block?

nragon, 2017-02-21:
I made a pull request with this issue. It's an idea which might do. But I'm not able to test it yet. I believe this issue happens because `IoDisk` uses `imwrite` to save to disk but because `Image` is still in scope when saving the next Io's the reference to `cv::Mat` which was used to save the image is locking that file.

cedricve, 2017-02-21:
I don't see any issue with this.. Can you explain why you think this is blocking? Though please create a pull request, after you've tested it properly.

nragon, 2017-02-21:
Sorry about the pull, but I didn't have the time to set the environment to test.
After using `IoDisk` I'm using `IoScript`. Inside i'm runnig `run.sh` and calling a python script which tries to open the image sent as stdin. 
```
with open(image, 'rb') as f:
        data = f.read()
```
Can you confirm this snippet works in your side?
I can execute the script standalone but not through kerberosio.

Thanks

cedricve, 2017-02-21:
Thanks for clarification, did you tried to log to a file?

nragon, 2017-02-21:
Yes, everything happens correctly on kerberosio side (Motion detected -> IoDisk -> IoScript) until it tries to open the file in the script. I've log untill `open()` and inside it but with not success.
In order for you to test, if possible.
run.sh:
```
#!/bin/bash

# -------------------------------------------
# This is an example script which illustrates
# how to use the Script IO device.
#

# --------------------------------------
# The first parameter is the JSON object
#
# e.g. {"regionCoordinates":[308,250,346,329],"numberOfChanges":194,"timestamp":"1486049622","microseconds":"6-161868","token":344,"pathToImage":"1486049622_6-161868_frontdoor_308-250-346-329_194_344.jpg","instanceName":"frontdoor"}

scriptDir=`dirname "$0"`
scriptDir=`cd "${scriptDir}"; pwd`

JSON=$1

# -------------------------------------------
# You can use python to parse the JSON object
# and get the required fields

name=$(echo $JSON | python -c "import sys, json; print json.load(sys.stdin)['instanceName']")
coordinates=$(echo $JSON | python -c "import sys, json; print json.load(sys.stdin)['regionCoordinates']")
changes=$(echo $JSON | pythfon -c "import sys, json; print json.load(sys.stdin)['numberOfChanges']")
image=$(echo $JSON | python -c "import sys, json; print json.load(sys.stdin)['pathToImage']")

python "${scriptDir}/kerberos.py" $image
```

On python script just try to open the catured image:

```
import sys

capture = sys.argv[1]
with open(capture , 'rb') as f:
        data = f.read()
        print "Works"
```

I don't have the scripts with me atm, if this doesn't work i'll post them later.
Thanks

cedricve, 2017-02-21:
Well you should add the directory path in front of `$image`.
By default only the name of the image is in the json object.

    # e.g. {"regionCoordinates":[308,250,346,329],"numberOfChanges":194,"timestamp":"1486049622","microseconds":"6-161868","token":344,"pathToImage"**:"1486049622_6-161868_frontdoor_308-250-346-329_194_344.jpg"**,"instanceName":"frontdoor"}

nragon, 2017-02-21:
Ok, `pathToImage` mistaken me. Closing pull. But it would be better to send full path.

cedricve, 2017-02-21:
Well not really because one can choose to access the image over URL (http://192.168.0.11/image.jpg) or over filesystem (/etc/opt/kerberosio/capture/image.jpg).

nragon, 2017-02-21:
Yes, true, either ways should have the prefix to image. Configurations should happen on kerberosio and not  hardcoded in scripts, imo.

cedricve, 2017-02-21:
What would you suggest @nragon ?

nragon, 2017-02-21:
Well, considering a user might want to send the image url thought script or access it directly it's better to add both options as attributes.
```
{
"regionCoordinates":[308,250,346,329],
"numberOfChanges":194,
"timestamp":"1486049622",
"microseconds":"6-161868",
"token":344,
"pathToImage": "/etc/opt/kerberosio/capture",
"urlToImage": "http://192.168.0.11",
"imageName":"1486049622_6-161868_frontdoor_308-250-346-329_194_344.jpg",
"instanceName":"frontdoor"
}
```

cedricve, 2017-02-21:
That could work for the filesystem, but not for the url. The machinery can run without the web interface, and doesn't have any knowledge (dependency) with the web interface.

cedricve, 2017-02-21:
Yeah you got a point, but when working with docker containers this will not work. I think it would be good to start with the pathToImage. Thanks for your feedback!

nragon, 2017-02-21:
No problem :)

closed 2017-02-22 by cedricve