kerberos-io/machinery #76: OpenCV : can't open url of ip camera

emarr, 2017-03-12:
I'm completely new to kerberos.io and am trying to get the machinery to take a stream from an IP camera.

I've updated config.xml...
`<capture file="capture.xml">IPCamera</capture>`

And for the purposes of testing am using the default URL provided in capture.xml...
```
    <IPCamera>
        <url type="text">rtsp://admin:888888@192.168.0.13/tcp/av0_1</url>
        <frameWidth type="number">640</frameWidth>
        <frameHeight type="number">360</frameHeight>
        <delay type="number">1000</delay>
        <angle type="number">0</angle>
    </IPCamera>
```

When I start kerberosis I get the following error
`[trivial] OpenCV : can't open url of ip camera`

Using tcpdump I can see that kerberos never tries to connect to 192.168.0.13 so something is wrong in the machinery.  I've tried setting...
`        <logging type="bool">true</logging>`
...but this doesn't show any additional information

I'm using kerberos.io built from source on Ubuntu 16.10

cedricve, 2017-03-12:
Is the ip camera on ip 192.168.0.13? this is the default value.

emarr, 2017-03-12:
No, my camera is on a different IP address but it doesn't matter URL I use in capture.xml, the machinery never tries to connect to it.

cedricve, 2017-03-12:
@emarr please try it with a valid ip, and report back. We are running a myriad of kerberos.io instances which processes an ip camera..

emarr, 2017-03-12:
To be honest, I tried it with a valid IP address at first and it didn't work so I reverted to the default configuration file to eliminate any chance that I had done something wrong.

The point is, when I start kerberosio, it doesn't make any attempt to establish a network connection to the IP address in the config file.  Even if the IP address is wrong, it should still try and connect.

cedricve, 2017-03-12:
Can you give me some system information? How you installed it and on which machine.

emarr, 2017-03-12:
It's a 64-bit installation of Ubuntu Server 16.10 running inside a VMware ESXi virtual machine.
`Linux ubuntu-builder 4.8.0-41-generic #44-Ubuntu SMP Fri Mar 3 15:27:17 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

I installed using the instuctions at https://doc.kerberos.io/2.0/installation/installation/generic

There were several warnings during the build - mainly relating to the use of teminating tokens (;) in the `#include` directives but `make check` and `make install` didn't complain.

When I run `kerberosio`:

```
12/03/2017 15:23:13.764 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
12/03/2017 15:23:13.764 INFO  [trivial] Parameters passed from commandline:
12/03/2017 15:23:13.764 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
12/03/2017 15:23:13.765 INFO  [trivial] Final configuration:
- algorithm = DifferentialCollins
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 1
- algorithms.BackgroundSubtraction.shadows = false
- algorithms.BackgroundSubtraction.threshold = 10
- algorithms.DifferentialCollins.erode = 5
- algorithms.DifferentialCollins.threshold = 15
- capture = IPCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 1000
- captures.IPCamera.frameHeight = 360
- captures.IPCamera.frameWidth = 640
- captures.IPCamera.url = rtsp://admin:888888@192.168.0.13/tcp/av0_1
- captures.RaspiCamera.angle = 0
- captures.RaspiCamera.delay = 500
- captures.RaspiCamera.frameHeight = 720
- captures.RaspiCamera.frameWidth = 1280
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.frameHeight = 640
- captures.USBCamera.frameWidth = 800
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 640
- captures.VideoCapture.frameWidth = 800
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket =
- clouds.S3.folder =
- clouds.S3.privateKey =
- clouds.S3.publicKey =
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 5000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- expositor = Hull
- expositors.Hull.region = 779,588|781,28|588,48|377,31|208,63|32,45|33,625|191,591|347,600|456,572|556,601|659,629
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 2
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Disk
- ios.Disk.directory = /etc/opt/kerberosio/capture/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = false
- ios.Disk.privacy = false
- ios.Disk.timestampColor = white
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
- ios.TCPSocket.message = motion detected
- ios.TCPSocket.port = 1337
- ios.TCPSocket.server = 127.0.0.1
- ios.Video.codec = h264
- ios.Video.directory = /etc/opt/kerberosio/capture/
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 3
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 120
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.timestampColor = white
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = true
- name = frontdoor
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 5
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- timezone = Europe-London
12/03/2017 15:23:13.765 INFO  [trivial] Logging is set to verbose
12/03/2017 15:23:13.765 INFO  [trivial] Starting cloud service: S3
12/03/2017 15:23:13.765 INFO  [trivial] Starting capture device: IPCamera
12/03/2017 15:23:13.766 ERROR [trivial] OpenCV : can't open url of ip camera
```

cedricve, 2017-03-12:
Thanks, do you have the compile log of OpenCV? To process IP camera's, ffmpeg needs to be compiled, I think the installation missed this.

emarr, 2017-03-12:
Where would I find the compile log?  I can rebuild from scratch if required - I have a VM snapshot taken before I started messing with kerberos.io

emarr, 2017-03-12:
FYI:
```
ffmpeg version 3.0.7-0ubuntu0.16.10.1 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 6.2.0 (Ubuntu 6.2.0-5ubuntu12) 20161005
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.10.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librubberband --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-chromaprint --enable-libx264
  libavutil      55. 17.103 / 55. 17.103
  libavcodec     57. 24.102 / 57. 24.102
  libavformat    57. 25.100 / 57. 25.100
  libavdevice    57.  0.101 / 57.  0.101
  libavfilter     6. 31.100 /  6. 31.100
  libavresample   3.  0.  0 /  3.  0.  0
  libswscale      4.  0.100 /  4.  0.100
  libswresample   2.  0.101 /  2.  0.101
  libpostproc    54.  0.100 / 54.  0.100
```

cedricve, 2017-03-12:
Ok, rebuilding will be the best option. If you can send the beginning of the opencv compilation.

e.g.

--   Video I/O:
--     DC1394 1.x:                  NO
--     DC1394 2.x:                  YES (ver 2.2.1)
--     FFMPEG:                      YES
--       codec:                     YES (ver 56.1.0)
--       format:                    YES (ver 56.1.0)
--       util:                      YES (ver 54.3.0)
--       swscale:                   YES (ver 3.0.0)
--       resample:                  YES (ver 2.1.0)
--       gentoo-style:              YES

emarr, 2017-03-12:
[putty-ubuntu-builder-20170312154517.txt](https://github.com/kerberos-io/machinery/files/836618/putty-ubuntu-builder-20170312154517.txt)

I've attached a complete transcript of the build, install and first run of kerberos.io.  This was on a clean system.

I changed 2 things from the instructions:
No sudo for `git clone https://github.com/kerberos-io/machinery`
`make -j 8` to use more CPU cores

cedricve, 2017-03-12:
Thanks, well ffmpeg isn't found. This is the reason why it's not polling.

--     DC1394 2.x:                  NO
--     FFMPEG:                      NO
--       avcodec:                   NO
--       avformat:                  NO
--       avutil:                    NO
--       swscale:                   NO
--       avresample:                NO

Try to install ffmpeg again, and rebuild.
http://ubuntuhandbook.org/index.php/2016/09/install-ffmpeg-3-1-ubuntu-16-04-ppa/

emarr, 2017-03-12:
Is ffmpeg 3.2 necessary?  I already have 3.0.7 as per my comment above.
I ask, because the ppa you suggested doesn't include a build for yakkety.
Trying to build latest ffmpeg from source alse has me entering a world of pain,

cedricve, 2017-03-23:
Hey @emarr  any update on this?

emarr, 2017-03-25:
I abandoned getting this installed directly under Ubuntu 16.10 and went down the Docker route..  As I say, I do have ffmpeg installed so not sure why OpenCV isn't picking that up.  Trying to build ffmpeg from source was a royal pain in the arse, mainly because of the tree of dependencies that needed to be built first..

closed 2017-03-25 by cedricve

JanSimek, 2017-07-27:
Perhaps you should add a simple check for FFMPEG in your machinery CMake script or maybe the web interface. I had the same problém and it took me a while before I found this bug report. I know instructions mention FFMPEG, but this would be a one line patch.

cedricve, 2017-07-28:
@JanSimek if you have time to add a PR? would be kind.

JanSimek, 2017-07-29:
Oh nevermind, I thought there was a CMake find module for FFMPEG, but there isn't. I still think it would be worth it to add it, for example from Kitware: https://github.com/Kitware/VTK/blob/master/CMake/FindFFMPEG.cmake

or copy the ```ocv_check_modules``` macro from OpenCV itself and do it like they do using pkg-config. However, even after installing FFMPEG, I could't get Kerberos running correctly (https://github.com/kerberos-io/web/issues/87) so I will have to try something else

Naprosnia, 2017-07-29:
i got the same problem. and realize that when we install ffmpeg with prefix /usr/local compiling opencv find it, and when we use prefix /usr opencv cannot find.

There's a way to compile opencv and make it look at /usr instead os /usr/local?
I installed ffmpeg from ded-multimedia.org and it uses prefix /usr.

cedricve, 2017-07-29:
@JanSimek this might just be a health-check issue. If you have the status command showing it's ok than you're good. Maybe you can try out Docker instead https://doc.kerberos.io/2.0/installation/Docker, so you don't have to mess with the dependencies.