kerberos-io/machinery #77: After a few days connection to USB camera lost (VIDIOC_DQBUF: No such device)

mishab, 2017-03-21:
I notice this issue a few month ago when after a while I get the "VIDIOC_DQBUF: No such device" error. I thought the issue was related to the the USB camera losing power (it was powered by the PI), but I also got it even after using the external powered USB hub.

When trying to restart the app I get the following error:
- streams.Mjpg.enabled = true
- streams.Mjpg.quality = 100
- streams.Mjpg.streamPort = 8888
- timezone = America-Los_Angeles
21/03/2017 21:48:13.489 INFO  [trivial] Logging is set to info
21/03/2017 21:48:13.490 INFO  [trivial] Starting cloud service: S3
21/03/2017 21:48:13.493 INFO  [trivial] Starting capture device: USBCamera
VIDEOIO ERROR: V4L: index 0 is not correct!
21/03/2017 21:48:13.816 INFO  [trivial] Stream: Configured stream on port 8888 with quality: 100
21/03/2017 21:48:15.317 INFO  [trivial] Starting conditions: Enabled
21/03/2017 21:48:15.317 INFO  [trivial] Starting algorithm: DifferentialCollins
21/03/2017 21:48:15.321 INFO  [trivial] Starting expositor: Hull
21/03/2017 21:48:16.944 INFO  [trivial] Starting heuristic: Sequence
21/03/2017 21:48:16.944 INFO  [trivial] Starting io devices: Disk
**OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /home/pi/machinery/build/opencv/modules/imgproc/src/color.cpp, line 8000
21/03/2017 21:48:16.983 ERROR [trivial] OpenCV : /home/pi/machinery/build/opencv/modules/imgproc/src/color.cpp:8000: error: (-215) scn == 3 || scn == 4 in function cvtColor**

I need to restart the PI in order to get machinery to function again. 
Any idea why this is happening?

cedricve, 2017-03-23:
Which camera are you using?  What is the resolution you're recording?

cedricve, 2017-04-26:
@mishab  please comment if you would like to add additional info. closing for now..

closed 2017-04-26 by cedricve