kerberos-io/machinery #78: Is `/etc/opt/` a valid installation path for all package contents?

agorf, 2017-04-19:
I've never seen `/etc/opt/` used as an installation path for all package contents. It is valid for configuration files etc. From what I've seen, `/opt/` is more frequently used to host a package installation as a whole.

carriba, 2017-04-23:
I have to second on the above: a program that just needs one directory and will get all files/libraries relative to that directory can get one directory for itself in **/opt/**. Maybe this can be amended in the next release? 

For more info, exemplary have a look in the article "[What is the best place to install user apps?](https://askubuntu.com/questions/1148/what-is-the-best-place-to-install-user-apps)", and other similar articles provide the same reasoning for the bespoken approach.

cedricve, 2017-04-23:
Thanks for your feedback guys, let's move it to /opt.

KarlBaumann, 2020-02-13:
I guess you never moved it anywhere, did you?
I think the worst is that it is creating logs file there instead of /var/log

cedricve, 2020-02-14:
indeed still not moved.