kerberos-io/machinery #81: Bad quality image from USB webcam after RPi reboot

dmnBrest, 2017-04-25:
Hello.

I have just started with your solution and it is GREAT! I like your Strategy! It is exactly what I'm looking for. Thanks for your work.

But I have one issue so far. I have I suppose default configuration: PRi + Logitech C270 USB webcam. All works fine after installation and during testing. But when I reload PRi the image from cam is some king noisy and motion detection now working. It can be fixed only if I open Configuration and just click the Update Button. 

Here are screenshot right after RPi reboot and after mentioned "fix".
https://snag.gy/mEYc70.jpg
https://snag.gy/0PAxse.jpg

Where can be the problem? I can't even describe this issue to try to find it in google.

Thanks!

cedricve, 2017-04-25:
Which Raspberry Pi are you using, never heared from this before. Could be something related to your power supply, but though it's weird that when you restart it all works fine again. Did you tried another USB camera as well? Just to rule out the camera.

dmnBrest, 2017-04-25:
I use Raspberry Pi 2 Model B.
Didn't try another Web cam yet, but I will try soon.
And let you know.

cedricve, 2017-04-26:
Ok great @dmnBrest can you reopen the issue if you have more info?

closed 2017-04-26 by cedricve