kerberos-io/machinery #82: USB/IP Cams

dmnBrest, 2017-04-25:
I want to build surveillance system for my off town house. 
I started with my available USB web cams + RPi but as I see results are not so good.
The angle of most webcams is not enough. I investigated this question and it seems there just few webcams that have wide angle but they are not so cheap. Also night vision not presented in USB webcams. So I have 2 questions or maybe I want to ask your advice.
What usb web cams are the best solution for surveillance system?
Is there a way to connect IP Cam directly to RPi? I want to minimize my solution and I don't want to use any routers. Just Cam+RPi+GSM Dongle.

cedricve, 2017-04-25:
hey @dmnBrest, I'm happy with my Logitech C270. It has been running 24/7 for two years now, and still not broken; this is crazy for only 25$. If you want to have a IR high quality camera you probably are looking for an expensive ip camera (200-600€), you might consider if that's useful if you want to have a minimum solution.

We didn't test Kerberos.io with a GSM dongle so far, so can't guide you on this..

Cédric

dmnBrest, 2017-04-25:
As for me I not so happy with my Logitech C270 - it don't cover any room - just a small peace. And I don't want to pay more than 100$ for complete solution. I plan to have cheaper solution to use it in far-away places. 
Did you test kerberos for Raspberry Zero?

cedricve, 2017-04-26:
@dmnBrest yes, but for the moment Raspberry Pi Zero W isn't released. We are testing it https://github.com/kerberos-io/kios/issues/8. The regular Raspberry Pi

stickystyle, 2017-04-29:
While I haven't used it with kerberos yet, I've used PlayStation Eye camera's in other surveillance tools. With a 75 degree FOV, long USB cable and the fact you can find them used for $5 I think they are pretty good.

dmnBrest, 2017-04-30:
If I'm not mistaken PlayStation Eye camera has 640x480. Is it enough for surveillance system, furthermore for wide angle? Can you upload somewhere (ex: https://snag.gy/) examples of photos and share link here?

stickystyle, 2017-05-02:
That's correct, 640x480. 
In my use case I'm just looking to identify people coming through the door and see which pet is messing with the blinds, don't really need any more pixels than 640x480 for that. The shot I have here isn't the greatest example since it's pointing at a sun-lit window, but you can get the overall idea. The wall is approximately 12 feet away I've tested resolution to be enough to read letters that are 2" tall at that distance.

https://snag.gy/UIczrS.jpg

closed 2017-07-25 by cedricve