kerberos-io/machinery #83: Optimize computer vision performance on single core boards

cedricve, 2017-04-26:
Kerberos.io works "fine" on Raspberry Pi 2 and 3. When recording video we can achieve a framerate of 7-10FPS at a resolution of 640x480. On a Raspberry Pi Zero (W) or Raspberry Pi 1, the framerate is at average 3FPS.

# Functions to optimize
With this issue we want to tackle different parts of the machinery, that may be improved. Most of them are related to standard OpenCV functions.

1. Differential images, calculating the changed pixels between frames.
https://github.com/kerberos-io/machinery/blob/master/src/kerberos/machinery/algorithm/DifferentialCollins.cpp#L28-L40

2. Expositor, iterating over pixels to find the changed pixels within a ROI.
https://github.com/kerberos-io/machinery/blob/master/src/kerberos/machinery/expositor/RectangleExpositor.cpp#L31-L44

3. Encoding MJPEG streaming, for viewing a live stream from the web interface.
https://github.com/kerberos-io/machinery/blob/master/src/kerberos/capture/Stream.cpp#L167

4. Video writing with FFMPEG (x264)
https://github.com/kerberos-io/machinery/blob/master/src/kerberos/machinery/io/IoVideo.cpp#L300-L323

muten84, 2017-07-22:
Hi any chance to get it work with IPCamera too? I'm interested in video recording from IPCamera.

cedricve, 2017-07-22:
it's possible for IP camera's also. The idea is instead of using OpenCV to write video files (which is slow and heavy operation), we'll fire up an FFMPEG command with omx hardware encoder. This will give same performance as for the Raspberry Pi camera (30FPS).

Problem that needs to be solved is compiling the omx libraries properly on Raspberry Pi (KiOS).
https://github.com/ccrisan/motioneyeos/pull/998

muten84, 2017-07-22:
I'm going to check [this](https://www.reddit.com/r/raspberry_pi/comments/5677qw/hardware_accelerated_x264_encoding_with_ffmpeg/ ) reddit. However did you think Raspbian users of Kerberos.io can just use video hardware accelerated video encoding??

cedricve, 2017-07-22:
Yes, but you'll need to adjust the IoVideo class to use FFMPEG instead of OpenCV.

muten84, 2017-07-22:
i will give it a try using a command line pipe from live stream and ffmpeg encoding .... if it works the next step could be to use the pipe for writing images and then start ffmpeg encoding that reads from the just created pipe. I didn't remember how to use pipes or shared memory in c++ but i think this is possible

douggb, 2017-09-13:
If you need people to test, I have a Pi2, Pi3 and an OrangePi+2e. IP camera Hikvision 4mp. Trying to get 1080p at 15fps. Still learning Linux, so would need some guidance.

cedricve, 2017-11-04:
###Update on hardware encoding

Started experimenting with modifying the OpenCV video writer class. Hardcoded the h264_omx codec. https://github.com/cedricve/opencv/blob/h264-preset/modules/videoio/src/cap_ffmpeg_impl.hpp#L1458-L1462. Encoding is successful on 800 x 600 and 1920x1080, however doesn't work on other resolutions (600x480 and 1280x720). Need to investigate why. @jasaw reported a similar issue in his PR for the motioneyeos project. ccrisan/motioneyeos#365

cedricve, 2017-11-15:
Successfully encoding with h264_omx codec, for USB and IP cameras, after upgrading to FFMPEG 3.4.3 and removing zero_copy with patch (ccrisan/motioneyeos#998). Thanks to @jasaw.