kerberos-io/machinery #86: questions about the configuration

denniis246, 2017-05-30:
Hello,

i use a raspberrypi 3, a raspicam and 2 IP-Cameras.
i followed the installation guide on https://doc.kerberos.io/2.0/installation/Raspbian
after this i saw my raspicam in the webinterface and everything worked.
next step was to include my 2 IP-Cameras: https://doc.kerberos.io/2.0/installation/Multi-camera/Raspbian

i did:

config/io.xml -> codec type changed to h264
captures.xml -> change ip to rtsp://USERNAME:PASSWORT@IP:554/mpeg4 (works von VLC)

tryed: 
kerberosio --name ipcamera1 --capture IPCamera --captures.IPCamera.deviceNumber 0 --streams.Mjpg.enabled false

it did not work. can you explain me what my mistake is?

and an other question, is it possible to view the 2 rtsp streams of my ip cameras on the webinterface?
i was reading that i have to change the ports to even activate the stream, can you explain me how? is it also a parameter in the commandline?
pictures and videos from the IP- Cameras would be stored in /etc/opt/kerberosio/capture and so they would be viewable as event of the webinterface, am i right?

sorry for my bad english
this is a great video surveillance solution tool and i would love to master it!

best regards
dennis

denniis246, 2017-05-30:
to the question with the livestream

--name ipcamera1 --capture IPCamera --captures.IPCamera.deviceNumber 0 --captures.RaspiCamera.frameHeight 720 -- captures.RaspiCamera.frameWidth 640 --captures.IPCamera.url  -- rtsp://admin:123456@192.168.178.125:554/mpeg4 --streams.Mjpg.enabled true --streams.Mjpg.streamport 8890

could this work?

cedricve, 2017-05-30:
hey @denniis246,

Can you try to use this url "rtsp://admin:123456@192.168.178.125:554/mpeg4?", before configuring multi-camera make sure you do the regular configuration.

Cédric

denniis246, 2017-05-30:
so i should disable my raspicam and only use the IP camera?

denniis246, 2017-05-30:
Hello Cédric,

what i did:
config.xml -> capture to IPCamera
capture.xml -> url to rtsp://admin:123456@192.168.178.125:554/1/mpeg4

restart machinery

what i got:

root@raspberrypi:/etc/opt/kerberosio/config# kerberosio
30/05/2017 18:38:39.045 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
30/05/2017 18:38:39.045 INFO  [trivial] Parameters passed from commandline:
30/05/2017 18:38:39.045 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
30/05/2017 18:38:39.048 INFO  [trivial] Final configuration:
- algorithm = DifferentialCollins
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 1
- algorithms.BackgroundSubtraction.shadows = false
- algorithms.BackgroundSubtraction.threshold = 10
- algorithms.DifferentialCollins.erode = 5
- algorithms.DifferentialCollins.threshold = 15
- capture = IPCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 500
- captures.IPCamera.frameHeight = 480
- captures.IPCamera.frameWidth = 640
- captures.IPCamera.url = rtsp://admin:123456@192.168.178.125:554/1/mpeg4
- captures.RaspiCamera.angle = 0
- captures.RaspiCamera.delay = 500
- captures.RaspiCamera.frameHeight = 480
- captures.RaspiCamera.frameWidth = 640
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.frameHeight = 480
- captures.USBCamera.frameWidth = 640
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 480
- captures.VideoCapture.frameWidth = 640
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket = 
- clouds.S3.folder = 
- clouds.S3.privateKey = 
- clouds.S3.publicKey = 
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 5000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- expositor = Hull
- expositors.Hull.region = 779,588|781,28|588,48|377,31|208,63|32,45|33,625|191,591|347,600|456,572|556,601|659,629
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 2
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Video
- ios.Disk.directory = /etc/opt/kerberosio/capture/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = false
- ios.Disk.privacy = false
- ios.Disk.timestampColor = white
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
- ios.TCPSocket.message = motion-detected
- ios.TCPSocket.port = 1337
- ios.TCPSocket.server = 127.0.0.1
- ios.Video.codec = h264
- ios.Video.directory = /etc/opt/kerberosio/capture/
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 3
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 120
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.timestampColor = white
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = false
- name = raspicam
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 5
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- timezone = Europe-Brussels
30/05/2017 18:38:39.050 INFO  [trivial] Logging is set to info
30/05/2017 18:38:39.050 INFO  [trivial] Starting cloud service: S3
30/05/2017 18:38:39.051 INFO  [trivial] Starting capture device: IPCamera
[rtsp @ 0x2351440] Missing packets; dropping frame.
[rtsp @ 0x2351440] Missing packets; dropping frame.
[rtsp @ 0x2351440] Missing packets; dropping frame.
[rtsp @ 0x2351440] Missing packets; dropping frame.
[rtsp @ 0x2351440] Missing packets; dropping frame.
[rtsp @ 0x2351440] Missing packets; dropping frame.

further advices?

best regards
Dennis

denniis246, 2017-05-30:
no, /etc/opt/kerberosio/capture is empty.
i noticed that the rtsp stream in vlc has a 10 seconds delay.
it also freezes after a few minutes in the vlc player.

tested the stream on windows -> same delay but no freeze.
do i miss any codec packs?

denniis246, 2017-05-30:
update @cedricve 

i changed url to rtsp://192.168.178.125:554/mpeg4

the IPCamera was shortly shown in the webinterface and i got captures.
but it aborted after 1 or 2 miinutes

got:

pi@raspberrypi:/etc/opt/kerberosio/config $ kerberosio
31/05/2017 00:07:42.200 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
31/05/2017 00:07:42.202 INFO  [trivial] Parameters passed from commandline:
31/05/2017 00:07:42.202 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
31/05/2017 00:07:42.219 INFO  [trivial] Final configuration:
- algorithm = DifferentialCollins
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 1
- algorithms.BackgroundSubtraction.shadows = false
- algorithms.BackgroundSubtraction.threshold = 10
- algorithms.DifferentialCollins.erode = 5
- algorithms.DifferentialCollins.threshold = 15
- capture = IPCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 500
- captures.IPCamera.frameHeight = 480
- captures.IPCamera.frameWidth = 640
- captures.IPCamera.url = rtsp://admin:123456@192.168.178.125:554/mpeg4
- captures.RaspiCamera.angle = 0
- captures.RaspiCamera.delay = 500
- captures.RaspiCamera.frameHeight = 480
- captures.RaspiCamera.frameWidth = 640
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.frameHeight = 480
- captures.USBCamera.frameWidth = 640
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 480
- captures.VideoCapture.frameWidth = 640
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket = 
- clouds.S3.folder = 
- clouds.S3.privateKey = 
- clouds.S3.publicKey = 
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 5000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- expositor = Hull
- expositors.Hull.region = 779,588|781,28|588,48|377,31|208,63|32,45|33,625|191,591|347,600|456,572|556,601|659,629
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 2
[... 226 lines cut ...]
- ios.Video.directory = /etc/opt/kerberosio/capture/
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 3
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 120
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.timestampColor = white
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = false
- name = frontdoor
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 5
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- timezone = Europe-Brussels
31/05/2017 00:13:11.721 INFO  [trivial] Logging is set to info
31/05/2017 00:13:11.722 INFO  [trivial] Stopping cloud service
31/05/2017 00:13:11.722 INFO  [trivial] Starting cloud service: S3
31/05/2017 00:13:11.723 INFO  [trivial] Stopping streaming
31/05/2017 00:13:11.723 INFO  [trivial] Stream: Succesfully closed streaming
31/05/2017 00:13:11.739 INFO  [trivial] Stopping capture device
Segmentation fault

cedricve, 2017-05-30:
Hmm not good, which IP camera's do you have?

denniis246, 2017-05-31:
i use hd ip camera from dongjia model dj-ipc-hd6138hdv

denniis246, 2017-05-31:
Hi @cedricve 

i just updated my ipcamera:
Substream --> Endcode: H264, Resoloution CIF, Rate Controll CVBR, FPS (10) - default
new RTSP -> rtsp://admin:123456@192.168.178.125/mpeg4c

in the io.xml i changed under video the fps to 10 also.
now it works without issues.

here is what i got:
pi@raspberrypi:/etc/opt/kerberosio/config $ kerberosio
31/05/2017 12:45:23.990 INFO  [trivial] Logging is written to: /etc/opt/kerberosio/logs/log.stash
31/05/2017 12:45:23.991 INFO  [trivial] Parameters passed from commandline:
31/05/2017 12:45:23.991 INFO  [trivial] Reading configuration file: /etc/opt/kerberosio/config/config.xml
31/05/2017 12:45:23.994 INFO  [trivial] Final configuration:
- algorithm = DifferentialCollins
- algorithms.BackgroundSubtraction.dilate = 7
- algorithms.BackgroundSubtraction.erode = 5
- algorithms.BackgroundSubtraction.history = 15
- algorithms.BackgroundSubtraction.nmixtures = 5
- algorithms.BackgroundSubtraction.ratio = 1
- algorithms.BackgroundSubtraction.shadows = false
- algorithms.BackgroundSubtraction.threshold = 10
- algorithms.DifferentialCollins.erode = 5
- algorithms.DifferentialCollins.threshold = 15
- capture = IPCamera
- captures.IPCamera.angle = 0
- captures.IPCamera.delay = 500
- captures.IPCamera.frameHeight = 480
- captures.IPCamera.frameWidth = 640
- captures.IPCamera.url = rtsp://admin:123456@192.168.178.125:554/mpeg4cif
- captures.RaspiCamera.angle = 0
- captures.RaspiCamera.delay = 500
- captures.RaspiCamera.frameHeight = 480
- captures.RaspiCamera.frameWidth = 640
- captures.USBCamera.angle = 0
- captures.USBCamera.delay = 500
- captures.USBCamera.deviceNumber = 0
- captures.USBCamera.frameHeight = 480
- captures.USBCamera.frameWidth = 640
- captures.VideoCapture.angle = 0
- captures.VideoCapture.delay = 500
- captures.VideoCapture.frameHeight = 480
- captures.VideoCapture.frameWidth = 640
- captures.VideoCapture.path = 0
- cloud = S3
- clouds.S3.bucket = 
- clouds.S3.folder = 
- clouds.S3.privateKey = 
- clouds.S3.publicKey = 
- condition = Enabled
- conditions.Enabled.active = true
- conditions.Enabled.delay = 5000
- conditions.Time.delay = 10000
- conditions.Time.times = 0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59-0:01,23:59
- expositor = Hull
- expositors.Hull.region = 779,588|781,28|588,48|377,31|208,63|32,45|33,625|191,591|347,600|456,572|556,601|659,629
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
[... 124 lines cut ...]
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.timestampColor = white
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = false
- name = frontdoor
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 5
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- timezone = Europe-Brussels
31/05/2017 12:49:32.516 INFO  [trivial] Logging is set to info
31/05/2017 12:49:32.516 INFO  [trivial] Stopping cloud service
31/05/2017 12:49:32.517 INFO  [trivial] Starting cloud service: S3
31/05/2017 12:49:32.518 INFO  [trivial] Stopping streaming
31/05/2017 12:49:32.529 INFO  [trivial] Stream: Succesfully closed streaming
31/05/2017 12:49:32.529 INFO  [trivial] Stopping capture device
31/05/2017 12:49:37.939 INFO  [trivial] Starting capture device: IPCamera
31/05/2017 12:49:42.048 INFO  [trivial] Stream: Configured stream on port 8889 with quality: 75
31/05/2017 12:49:42.049 INFO  [trivial] Starting conditions: Enabled
31/05/2017 12:49:42.050 INFO  [trivial] Starting algorithm: DifferentialCollins
31/05/2017 12:49:42.050 INFO  [trivial] Starting expositor: Hull
31/05/2017 12:49:42.254 INFO  [trivial] Starting heuristic: Sequence
31/05/2017 12:49:42.254 INFO  [trivial] Starting io devices: Video

cedricve, 2017-05-31:
awesome @denniis246, do you need more help?

denniis246, 2017-05-31:
yeah, i have done the same substream configuration to the other IPCamera and would love to know wich parameters i need to use to get it working. (also with livestream)

and the same for my raspicam. 

best regards
dennis

cedricve, 2017-05-31:
well, you can configure the web interface to run with the RaspiCamera, and execute 2 other binaries to procress your 2 IP camera's.

E.g.

1. Configure web interface to use Raspicamera
2. Run kerberosio for first ip camera: `kerberosio --name ipcamera1 --capture IPCamera --captures.IPCamera.url  rtsp://admin:123456@192.168.178.125:554/mpeg4cif --streams.Mjpg.streamPort 8890 &`
2. Run kerberosio for second ip camera: `kerberosio --name ipcamera1 --capture IPCamera --captures.IPCamera.url  rtsp://admin:123456@ip-of-second-camera:554/mpeg4cif --streams.Mjpg.streamPort 8891 &`

denniis246, 2017-06-01:
Hello @cedricve ,

it seems to work now, is it possible to view the livestreams of the ip cameras in the webinterface also? and thank you very much for your time!

cedricve, 2017-06-02:
hey @denniis246, nope not in the interface. You can use tools like IP camera viewer though..

closed 2017-06-02 by cedricve