kerberos-io/machinery #87: PHP client

stgleb, 2017-07-03:
Is there any possibilities to contact with machinery in php? Existing clients or I need to prepare json with task by my own?

cedricve, 2017-07-04:
hey @stgleb, yes you can if you use the REST API of the web interface: https://doc.kerberos.io/2.0/web/aPI.
Find a complete list of commands here: https://github.com/kerberos-io/web/blob/master/app/api.php#L79-L118

If you feel the need to have more methods, feel free to contribute them (it's not that hard).
FYI: we'll create a new release by the end of this week (the structure of this repo will slightly change).

stgleb, 2017-07-04:
@cedricve Yes, but I don't need to use kerberos at all, just machinery for async tasks.

cedricve, 2017-07-04:
Well, the way the web communicates with the machinery is by modifying the config files in the `config/`directory. So you'll end up with this solution as well.

closed 2017-07-25 by cedricve