kerberos-io/machinery #88: NAS mount error

denniis246, 2017-07-06:
Hello Cedric,

im trying to store images directly to the nas.
but the installation guide wont work.

here is my config:

# file /etc/fstab
roc            /proc           proc    defaults          0       0
PARTUUID=a764ef18-01  /boot           vfat    defaults          0       2
PARTUUID=a764ef18-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
//192.168.217.30/Kamera /etc/opt/kerberosio/capture cifs username=pi,password=schueler,iocharset=utf8,file_mode=0777,dir_mode=0777 0 0
## file to run kerberos-io
!/bin/bash
#ZEITSERVER:
ntpdate ptbtime1.ptb.de
#hwclock -uw

sleep 15s

#NAS MOUNT:
#mount -t cifs //192.168.217.30/Kamera /NAS -o username=Kamera,password=schueler,uid=pi

#sleep 15s

#IP-ADRESSE:
ifconfig eth0 192.168.178.100/24
sleep 1s

#KERBEROSIO
kerberosio &
sleep 20s
kerberosio --name ipcamera1 --capture IPCamera --captures.IPCamera.url rtsp://admin:schueler@192.168.178.125:554/mpeg4cif --streams.Mjpg.streamPort 8891 &
sleep 8s
kerberosio --name ipcamera2 --capture IPCamera --captures.IPCamera.url rtsp://admin:schueler@192.168.178.126:554/mpeg4cif --streams.Mjpg.streamPort 8892 &
sleep 8s
kerberosio --name ipcamera3 --capture IPCamera --captures.IPCamera.url rtsp://admin:schueler@192.168.178.127:554/mpeg4cif --streams.Mjpg.streamPort 8893 &
sleep 8s
#kerberosio --name ipcamera4 --capture IPCamera --captures.IPCamera.url http://admin:schueler@192.168.178.128:80/videostream.cgi --streams.Mjpg.streamPort 8894 &

i am trying to run mount -a
issue:
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)

any advice?

denniis246, 2017-07-06:
issue solved src=ntlm

denniis246, 2017-07-06:
new issue, after i mount the nas the webinterface wont work.
error:
The 192.168.178.100 page isn’t working

192.168.178.100 redirected you too many times.
Try clearing your cookies.
ERR_TOO_MANY_REDIRECTS

clearing cookies wont work.

cedricve, 2017-07-12:
@denniis246  can I close this issue?

denniis246, 2017-07-12:
yes you canVon meinem Huawei-Mobiltelefon gesendet-------- Originalnachricht --------Betreff: Re: [kerberos-io/machinery] NAS mount error (#88)Von: Cédric Verstraeten An: kerberos-io/machinery Cc: denniis246 ,Mention @denniis246  can I close this issue?

closed 2017-07-12 by cedricve