kerberos-io/machinery #89: Provide an IO service for publishing events to MQTT

flaviostutz, 2017-07-17:
In the same way we have Webhook/TCP, there could be the possiblity of Kerberos.io to publish events to an MQTT broker. 

That way it could be much simpler for various engines to listen to "motion" events and perform various tasks in parallel. One could look for people crossing, other to people identification, other to car counting, other to store snapshots on the cloud, just like plugins, but with no need to change the Marchinery code, in parallel and over the network (each one of these features could be a separate running container, on site or at the cloud).

We could start with simple event publishing (just like Webhook/TCP implementation), but them there could be some sort of time aggregation (just like "video" does when recording 5s of footage etc).

flaviostutz, 2017-07-17:
If it is OK for you guys (it's a great work you're doing here!), I will try to start an implementation on this.

cedricve, 2017-07-17:
@flaviostutz amazing ideas, yeah would be great! do you need any more info?

cedricve, 2017-07-18:
@flaviostutz let me know if you need help.

cedricve, 2017-08-18:
If someone can help contributing on this? I would love to pair program this feature.

cedricve, 2017-08-21:
I've supplied a template for the MQTT Io device on the develop branch. Please find files here:

https://github.com/kerberos-io/machinery/blob/develop/include/kerberos/machinery/io/IoMQTT.h
https://github.com/kerberos-io/machinery/blob/develop/src/kerberos/machinery/io/IoMQTT.cpp

zingelst, 2017-09-08:
As a small suggestion, you may be able to achieve this using your existing Script IO plugin. Using the [mosquitto-clients](https://packages.debian.org/jessie/mosquitto-clients) package (which should exist on Raspberry Pi, it's in Debian already), you could easily make a script that would publish to a MQTT broker. I'm not quite sure how your configuration gets to the Script plugin, but you'd need the MQTT broker host IP along with any security/QOS/etc options. 

You'd at least be able to try it out without having to add code or an additional dependency (on say [libmosquitto](https://mosquitto.org/api/files/mosquitto-h.html) I do have some experience with libmosquitto. It's a very nice, small library which is pretty easy to use. If no one has gotten too far already, I might have some bandwidth to contribute some code to help out.

cedricve, 2017-09-19:
@zingelst would be great, let me know if you have some questions.

flaviostutz, 2017-09-19:
Unfortunately I had no time to code this yet. Sorry.
I was thinking it would publish the image on the topic too (not just metadata, configurable). Would it be too heavy?

cedricve, 2017-09-19:
Sorry publish the image on the topic?

zingelst, 2017-09-19:
MQTT can handle messages up to 256MB, images shouldnt be problem

cedricve, 2017-09-19:
aha ok, well I think videos shouldn't be a problem too but from my opinion you can just send a url instead of encoding the images in the message.

cedricve, 2017-10-22:
Hey guys,

@gianricod started working on a MQTT implementation, and we would like to have feedback. 

> @cedricve i have started working to it using libmosquittopp.
> 
> it is very basic at the moment.
> it sends a message to a topic upon each call of save(image,json). the message contains the json object. very similar to the tcpsocket in that sense.

    gianrico@vlin:~/machinery/build$ mosquitto_sub -h localhost -t "kios/mqtt" -v
    kios/mqtt {"regionCoordinates":[97,41,220,208],"numberOfChanges":483,"timestamp":"1508514764","microseconds":"6-471402","token":16}
    kios/mqtt {"regionCoordinates":[7,40,290,207],"numberOfChanges":1676,"timestamp":"1508514764","microseconds":"6-967244","token":16}

> 
> i see you have an enhancement open for that, we can discuss it there.
> it needs to be polished a bit with more options/settings (and maybe use cases as well).

cedricve, 2017-10-22:
UPDATE - FYI @gianricod, I've added a throttling function yesterday. Which helps with not flooding the MQTT service.  https://github.com/kerberos-io/machinery/blob/develop/src/kerberos/machinery/io/IoMQTT.cpp#L17

You can set the throttling rate in the configuration file (https://github.com/kerberos-io/machinery/blob/develop/config/io.xml#L43). So we can extend your current MQTT configuration with this also.

cedricve, 2017-10-30:
Update - @gianricon added a MQTT implementation. https://github.com/kerberos-io/machinery/pull/106.

I will be testing it in a node-red flow, would be nice to receive some feedback about possible usecases (real life scenarios).

cedricve, 2017-11-09:
Integrated with @gianricod PR: MQTT intitial implementation #106

closed 2017-11-09 by cedricve

wicadmin, 2018-09-11:
Use-case: Send to Telegram
Use-case: Send to Text-to-Speech in home automation/speaker system

I just implemented a node-red flow to send the alert and image URL to my Telegram bot and a messeage to my text-to-speech endpoint of home speakers. Thanks for the contribute!