kerberos-io/machinery #90: Provide a RTSP streaming with no recoding

flaviostutz, 2017-07-17:
In one of our projects we want to use Kerberos for advanced image processing, but for now we want to let video recording to the existing NVR infrastructure (it is well tested and people know how to use and operate them). So we thought we could connect the *only* cam feed to Kerberos.io and point the existing NVRs to the Kerberos (re)feed.

The majority of cams in the market can't perform more than one simultaneous RTSP/MJPEG feed on the cam itself (probably because of lack of onboard cpu power). On large deployments (hundreds), I don't want that kind of traffic duplication crowding the fibers too.

An important feature would be for the relay to perform restreaming with no recoding (maintaining original H264 encoded image from the camera) so that it could enable large scale deployments.

flaviostutz, 2017-07-17:
We created a RTSP relay (https://github.com/flaviostutz/rtsp-relay) that could be used as a reference based on FFMPEG. The problem with this is that it recodes the stream, consuming 50% CPU for each feed (seems like FFMPEG works this way).

We managed to relay RTSP on VLC (just on Windows!) with no recoding and it got 4% cpu overhead for each feed. Now we are trying to use GStreamer for this task (still on dependency hell epic)...

cedricve, 2017-07-17:
@flaviostutz nice, let's talk about this!

cedricve, 2017-08-15:
@flaviostutz maybe we should use OpenRTSP?

flaviostutz, 2017-08-16:
I read the documentation on http://www.live555.com/openRTSP/ but couldn't manage how to restream the feed. Could you send a sample command line for this?

Thanks.