kerberos-io/machinery #92: Build issue when installing on Ubuntu

anortiz08, 2017-07-26:
[100%] Linking CXX executable ../../test/bin/kerberos-test
[100%] Built target kerberos-test
[sudo] password for admin:
[  3%] Built target ENCODE_LIBRARY
[ 13%] Built target googletest
[ 23%] Built target restclient
[ 33%] Built target opencv
[ 40%] Built target TINYXML_LIBRARY
[ 42%] Built target EXECUTOR_LIBRARY
[ 50%] Built target FILEWATCHER_LIBRARY
[ 70%] Built target KERBEROS_CORE
[100%] Built target kerberos-test
Scanning dependencies of target check
../../test/bin/kerberos-test: error while loading shared libraries: libavcodec.so.56: cannot open shared object file: No such file or directory
test/CMakeFiles/check.dir/build.make:57: recipe for target 'test/CMakeFiles/check' failed
make[3]: *** [test/CMakeFiles/check] Error 127
CMakeFiles/Makefile2:1501: recipe for target 'test/CMakeFiles/check.dir/all' failed
make[2]: *** [test/CMakeFiles/check.dir/all] Error 2
CMakeFiles/Makefile2:1508: recipe for target 'test/CMakeFiles/check.dir/rule' failed
make[1]: *** [test/CMakeFiles/check.dir/rule] Error 2
Makefile:689: recipe for target 'check' failed
make: *** [check] Error 2

Naprosnia, 2017-07-26:
Instead of compile FFMPEG, install it like this- sudo apt-get install ffmpeg .

cedricve, 2017-07-27:
thanks for replying @Naprosnia! If you can try @anortiz08 his fix? How did you installed FFMPEG, did you also installed the shared libraries to your filesystem (a path which the dynamic library loader can read from)?

closed 2017-09-19 by cedricve

benoit34000, 2018-02-20:
Hi guys,
getting the same issue on  Debian 9 lxc container (Intel CPU)

I tried with and without ffmpeg (as Naporsnia proposal)

Is there any workaround ?
Thanks by advance,
Kind regards,
Benoit.

cedricve, 2018-02-20:
Which version of FFMPEG did you install? Can you find the libavcodec.so file on your filesystem?

benoit34000, 2018-02-20:
Hi,
1/ for ffmpeg version :
apt says :
ffmpeg/stable,now 7:3.2.10-1~deb9u1 amd64 [installed]

2/ for libavcodec.so :

/root/ffmpeg/libavcodec/libavcodec.so
/root/ffmpeg/libavcodec/libavcodec.so.56
/usr/lib/x86_64-linux-gnu/libavcodec.so.57
/usr/lib/x86_64-linux-gnu/libavcodec.so.57.64.101
/usr/local/lib/libavcodec.so
/usr/local/lib/libavcodec.so.56
/usr/local/lib/libavcodec.so.56.60.100

I think there are both apt package release version and the compiled ones from git source (regarding install guide)

Thanks for your help on this troubleshoot.
Kind regards,
Benoit.

jeremyroe, 2018-04-10:
Same issue occurring here.

rotragit, 2018-05-02:
try, after compiling ffmpeg:

sudo ldconfig

michal-f, 2018-10-28:
run command with ```sudo``` like : 

```
 cmake .. && sudo make && sudo make check && sudo make install
```

AnnoyingTechnology, 2018-11-09:
Thanks to @jeremyroe 
`ldconfig` did the trick for me on Debian 9.5.